dusty-nv/jetson-inference #957: I'm currently running this against 3 rtsp camera streams, and it works without a problem.

rlrahulkanojia, 2021-03-18:
I'm currently running this against 3 rtsp camera streams, and it works without a problem.
Every camera feed is read in a loop in a separate thread, and then I push images into a single `queue` for detectnet inference

_Originally posted by @tkislan in https://github.com/dusty-nv/jetson-inference/issues/855#issuecomment-753447378_

can someone help me with the implementation of the explanation using multiple stream In python

rlrahulkanojia, 2021-03-22:
@dusty-nv please can you help.

dusty-nv, 2021-03-22:
@rlrahulkanojia I haven't done it myself, this may be the queue object they are talking about:

https://docs.python.org/3/library/queue.html
https://docs.python.org/3/library/threading.html#thread-objects

Although using this multi-threading in Python may not even be necessary, because you can just poll multiple videoSource from the same thread using the Capture() timeout (as discussed in https://github.com/dusty-nv/jetson-inference/issues/960#issuecomment-804313584).  videoSource is already threaded underneath on the C++ side

DeepStream would probably give you better performance for multi-stream applications and is commonly used with multiple RTSP streams.

rlrahulkanojia, 2021-03-22:
Thanks! will also look into it

closed 2021-03-22 by rlrahulkanojia