dusty-nv/jetson-inference #958: after training model the onnx export fails

MST2010Markus, 2021-03-19:
jetson_nano@Jetson:~/jetson-inference/python/training/detection/ssd$ python3 onnx_export.py --model-dir=models/markusmodel
Namespace(batch_size=1, height=300, input='', labels='labels.txt', model_dir='models/markusmodel', net='ssd-mobilenet', output='', width=300)
running on device cuda:0
found best checkpoint with loss 0.760933 (models/markusmodel/mb1-ssd-Epoch-29-Loss-0.7609327760609713.pth)
creating network: ssd-mobilenet
num classes: 4
loading checkpoint: models/markusmodel/mb1-ssd-Epoch-29-Loss-0.7609327760609713.pth
Traceback (most recent call last):
File "onnx_export.py", line 86, in
net.load(args.input)
File "/home/jetson_nano/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 135, in load
self.load_state_dict(torch.load(model, map_location=lambda storage, loc: storage))
File "/home/jetson_nano/.local/lib/python3.6/site-packages/torch/nn/modules/module.py", line 1045, in load_state_dict
self.class.name, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SSD:
size mismatch for classification_headers.0.weight: copying a param with shape torch.Size([30, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 512, 3, 3]).
size mismatch for classification_headers.0.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).
size mismatch for classification_headers.1.weight: copying a param with shape torch.Size([30, 1024, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 1024, 3, 3]).
size mismatch for classification_headers.1.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).
size mismatch for classification_headers.2.weight: copying a param with shape torch.Size([30, 512, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 512, 3, 3]).
size mismatch for classification_headers.2.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).
size mismatch for classification_headers.3.weight: copying a param with shape torch.Size([30, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 256, 3, 3]).
size mismatch for classification_headers.3.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).
size mismatch for classification_headers.4.weight: copying a param with shape torch.Size([30, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 256, 3, 3]).
size mismatch for classification_headers.4.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).
size mismatch for classification_headers.5.weight: copying a param with shape torch.Size([30, 256, 3, 3]) from checkpoint, the shape in current model is torch.Size([24, 256, 3, 3]).
size mismatch for classification_headers.5.bias: copying a param with shape torch.Size([30]) from checkpoint, the shape in current model is torch.Size([24]).

dusty-nv, 2021-03-19:
Hi @MST2010Markus, I think this happens if the number of classes is different than the number of classes that the model was trained on.

Is the labels.txt file in your `models/markusmodel/` directory the one that was saved by train_ssd.py?

MST2010Markus, 2021-03-20:
Hi @dusty-nv , thanks , that helped, but now i got this:

jetson_nano@Jetson:~/jetson-inference/python/training/detection/ssd$ python3 onnx_export.py --model-dir=models/markusmodel
Namespace(batch_size=1, height=300, input='', labels='labels.txt', model_dir='models/markusmodel', net='ssd-mobilenet', output='', width=300)
running on device cuda:0
found best checkpoint with loss 10000.000000 ()
creating network:  ssd-mobilenet
num classes:       4
loading checkpoint:  models/markusmodel/
Traceback (most recent call last):
  File "onnx_export.py", line 86, in <module>
    net.load(args.input)
  File "/home/jetson_nano/jetson-inference/python/training/detection/ssd/vision/ssd/ssd.py", line 135, in load
    self.load_state_dict(torch.load(model, map_location=lambda storage, loc: storage))
  File "/home/jetson_nano/.local/lib/python3.6/site-packages/torch/serialization.py", line 571, in load
    with _open_file_like(f, 'rb') as opened_file:
  File "/home/jetson_nano/.local/lib/python3.6/site-packages/torch/serialization.py", line 229, in _open_file_like
    return _open_file(name_or_buffer, mode)
  File "/home/jetson_nano/.local/lib/python3.6/site-packages/torch/serialization.py", line 210, in __init__
    super(_open_file, self).__init__(open(name, mode))
IsADirectoryError: [Errno 21] Is a directory: 'models/markusmodel/'

dusty-nv, 2021-03-20:
What checkpoints do you have in your model dir?  It looks like it can't find one with a valid loss...are they NaN loss?

You can manually specify the full path to the pth checkpoint with the --input argument

closed 2023-04-27 by dusty-nv