dusty-nv/jetson-inference #959: too many resources requested for launch

lurending711, 2021-03-21:
2021-03-21 13:50:20 - Start training from epoch 0.
/usr/local/lib/python3.6/dist-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/usr/local/lib/python3.6/dist-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
THCudaCheck FAIL file=/local/jetbot/pytorch/aten/src/THCUNN/generic/SpatialDepthwiseConvolution.cu line=202 error=701 : **too many resources requested for launch**
Traceback (most recent call last):
  File "train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 123, in train
    loss.backward()
  File "/usr/local/lib/python3.6/dist-packages/torch/tensor.py", line 185, in backward
    torch.autograd.backward(self, gradient, retain_graph, create_graph)
  File "/usr/local/lib/python3.6/dist-packages/torch/autograd/__init__.py", line 127, in backward
    allow_unreachable=True)  # allow_unreachable flag
**RuntimeError: cuda runtime error (701) : too many resources requested for launch at /local/jetbot/pytorch/aten/src/THCUNN/generic/SpatialDepthwiseConvolution.cu:202**

dusty-nv, 2021-04-05:
Hi @lurending711, sorry for the delay - can you try reducing the batch size?

Also, are you using a PyTorch wheel you installed yourself, or one that the project installed or was in the container?  That 'too many resources requested for launch' issue is patched in the PyTorch wheels that I build, so unless there is a corner-case I haven't seen before, it shouldn't occur with those PyTorch wheels.   I also haven't seen this error on any of the Jetson's with train_ssd.py.

Have you changed the model's resolution to 512x512 or are you running it normally?

lurending711, 2021-07-05:
Thank you very much, I can already run normally. I didn't read your reply because of the exam. @dusty-nv

closed 2021-07-05 by lurending711

yahu-11, 2022-01-13:
> Thank you very much, I can already run normally. I didn't read your reply because of the exam. @dusty-nv
Hello, I also encountered such a problem. How did you solve it?

dogeheng, 2023-11-16:
> > Thank you very much, I can already run normally. I didn't read your reply because of the exam. @dusty-nv
> > Hello, I also encountered such a problem. How did you solve it?

Hi, @yahu-11 .I also have this problem,did you solve it?

dogeheng, 2023-11-16:
> Hi @lurending711, sorry for the delay - can you try reducing the batch size?
>
> Also, are you using a PyTorch wheel you installed yourself, or one that the project installed or was in the container? [...]

Hi, @dusty-nv .I have such problem too.And I have no idea how this happened.I have double check the resolution.It seems no problem to this.Have you figure out how this happened and is there a possible way to solve it?Thanks for your help.