dusty-nv/jetson-inference #960: Return Code in case pipeline stops

rlrahulkanojia, 2021-03-22:
I am calling the jetson infernece as a module but in case of any error, the pipeline just stops and does not return anything. is there a way to return something from the function which is calling the pipeline in case of EOS.

eg

def module(INPUT_URLS):
    input = jetson.utils.videoSource(INPUT_URLs[0], argv=sys.argv)]
    while True:
        frame = input.capture()
        ***
     (how to make sure this return something)
    return 200

rlrahulkanojia, 2021-03-22:
@dusty-nv  Please help me with this

dusty-nv, 2021-03-22:
@rlrahulkanojia, try the optional `timeout` argument to `videoSource.Capture()`, where timeout is in milliseconds.  By default, the timeout is -1, which means never timeout and wait indefinitely.  A timeout of 0 means return immediately if no image is ready.   A timeout of 1000 would mean wait 1 second for new image.

If a timeout occurs, it will throw a python exception, so you would want to catch that if you want your program to keep executing.

rlrahulkanojia, 2021-03-22:
Got it, Thanks!

closed 2021-03-22 by rlrahulkanojia