dusty-nv/jetson-inference #961: Object Detection Display Output

naimulhq, 2021-03-23:
We want to show the video output of the object detection in a GUI. What steps should we take to prevent the output window from popping up and instead output the video as a stream to our GUI (we are using Kivy)? If this is possible, what code/methods should we modify?

dusty-nv, 2021-03-23:
Hi @naimulhq, you can just skip the creation of the `videoOutput` object and not make an instance of that.  However if you still want it for recording/streaming, then you pass the `--headless` flag to the command-line args.  

* C++ - https://github.com/dusty-nv/jetson-inference/blob/6e078d21396298c3a9f1f1ea1f2c27bf80bbd4a6/examples/detectnet/detectnet.cpp#L72
* Python - https://github.com/dusty-nv/jetson-inference/blob/6e078d21396298c3a9f1f1ea1f2c27bf80bbd4a6/python/examples/detectnet.py#L55

naimulhq, 2021-03-28:
Hi @dusty-nv , I am trying to stream it and not trying to run the script through the command line. What I decided to do is 
`output = jetson.utils.videoOutput("display://0, ['--headless']) "`
However, it still creates the GL Display.

naimulhq, 2021-04-05:
I decided to just remove the variable and it ended up working. Thank you.

closed 2021-04-05 by naimulhq