dusty-nv/jetson-inference #962: multi models synchronize inference fps slow down

haige-algo, 2021-03-23:
I did a test on **jetson xavier**, there are two trt models, one is **detection** model, another is **freespace** model.

when infer detection model alone, it cost 20ms per frame.
when infer freespace model alone, it cost 30ms per frame.

then, I infer these two model at the same time, detection cost 35ms per frame average and freespace cost 43ms  per frame average. Two models both become slower than alone inference, it looks like when one model do inference another model have to wait. 

I am sure it not caused by GPU resource limitation, how to inference multi models synchronization, have any idea?

dusty-nv, 2021-03-23:
@haige-algo yes when two models are running on the GPU at same time, the GPU will need to task switch between them.  However it appears you have saved 7ms by running them in parallel, because you are done after 43ms (whereas before you were done after 50ms)

Since you are on Xavier, you can try running a model on DLA with TensorRT, because DLA is an independent resource.  There are two DLAs and GPU and they each run independently.

ismet-dagli, 2021-03-26:
Are you using GPU only? It can be related to GPU frequency. I strongly suggest watching GPU behavior while running single and multiple models.  
Xavier can change its mode under the different workloads.

closed 2023-04-27 by dusty-nv