dusty-nv/jetson-inference #964: segnet  makes shutdown Jetson Nano 4GB dev. kit

mipsan, 2021-03-24:
Very interesting. 

`imageNet `and `detectNet `work very well with both MIPI CSI camera and USB webcam on the Jetson Nano developer kit.
But `segNet `destroys the developer kit without any logs.

I increased **SWAP** space not to be starving.
```
jetson@jetson-desktop:~$ free
              total        used        free      shared  buff/cache   available
Mem:        4051072     1028068     2323216       38688      699788     2827924
Swap:      14680052           0    14680052
```
Where to dig please?

dusty-nv, 2021-03-24:
What is the terminal output when you run segnet?

mipsan, 2021-03-25:
> What is the terminal output when you run segnet?

Nothing at the debug console, it looks like instant power down.

`last` command shows like this:
```
jetson   pts/1        192.168.52.183   Thu Mar 25 10:24   still logged in
jetson   :0           :0               Thu Mar 25 10:19    gone - no logout
reboot   system boot  4.9.140-tegra    Sat Jan  1 10:00   still running
jetson   pts/1        192.168.52.183   Thu Mar 25 10:16 - crash (-7754+00:16)
jetson   :0           :0               Thu Mar 25 10:12 - crash (-7754+00:12)

```
V4L2 has following additional logs than MIPI CSI camera between Timing Reports:
```
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Constrained\ Baseline\ Profile\)", minimum-bitrate=(uint)3016080, maximum-bitrate=(uint)10865520, bitrate=(uint)4667240;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Constrained\ Baseline\ Profile\)", minimum-bitrate=(uint)2952240, maximum-bitrate=(uint)10865520, bitrate=(uint)4576976;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Constrained\ Baseline\ Profile\)", minimum-bitrate=(uint)2938320, maximum-bitrate=(uint)10865520, bitrate=(uint)4495044;
```

dusty-nv, 2021-03-25:
Does the board reboot or does the process just not run?   Does dmesg kernel log show anything?

mipsan, 2021-03-25:
Just turned off. And, unfortunately no kernel logs, but  `last` logs show simply '_crashed_'.

mipsan, 2021-03-25:
I cannot be sure following result is correct, just occurs on my desk.
I found two cases of crash, segnet and re-training.

Jetson Nano dev kit A02 is always crashed with both JetPack 4.3 and 4.5.
But it does not happen at B01 version at all.

Camera interfaces, memory and swap, I think, are not the issue at all.

dusty-nv, 2021-03-25:
If it is turning off, it may be a power supply issue.  Are you using the same power supply for both?

Try running this command:  `sudo nvpmodel -m 1` which will set the board to 5W mode.  Does it still shutdown when you run segnet?  If not, it was power supply issue.

mipsan, 2021-03-26:
Voila!

5W mode clears this issue!!
Thanks!

closed 2021-03-26 by mipsan

dusty-nv, 2021-03-26:
OK gotcha.  That means you would want to upgrade your power supply if you want to run it in 10W mode for best performance.