dusty-nv/jetson-inference #966: Getting camera window size

mhammaddaoud, 2021-03-25:
Hello,
When i run detectnet object detection, a window pops up displaying the camera with the detections boxes
How can i get the width and height of the display through python code?
Thank you.

dusty-nv, 2021-03-25:
Hi @mhammaddaoud, it should basically be the same as the camera's image width/height

mhammaddaoud, 2021-03-25:
Its changing according to the screen size

dusty-nv, 2021-03-25:
So the display takes up pretty much the whole screen?   It should match the camera image.

Do you want the dimensions of the screen or the window?  If you want the window dimensions, you will need to create a glDisplay object instead of videoSource, and then call `.GetWidth()` and `.GetHeight()` on it.  Here is older version of detectNet using glDisplay:

https://github.com/dusty-nv/jetson-inference/blob/20200714/python/examples/detectnet-camera.py

closed 2023-04-27 by dusty-nv