dusty-nv/jetson-inference #967: cmake

ZeeshanSar, 2021-03-25:
hi i just bought a jetson nano and trying get started with hello world. i am face a problem with the cmake the error is below.

### cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Run 'cmake --help' for more information.

how to fix this error?

ZeeshanSar, 2021-03-26:
i made i mistake while cmake ../ i just typed cmake

do i can add more images to the images folder to identify more objects?

dusty-nv, 2021-03-26:
Yes or you can save your images to a different directory and specify the full path to your image(s) that you want to process.

ZeeshanSar, 2021-03-26:
and how can i can run this remotely using IP address?

dusty-nv, 2021-03-26:
You would SSH into the device.  If you are on Windows, use an SSH client like Putty or MobaXTerm.

ZeeshanSar, 2021-03-26:
my jetson is giving an error for   ./detectne.console.py: no such file or directory. while testing

dusty-nv, 2021-03-26:
Is the working directory of your terminal `<path-to-your-jetson-inference>/build/aarch64/bin`

If not, go to that directory, or just run it as `detectnet.py` (no forward slash at beginning)

ZeeshanSar, 2021-03-26:
ok fixed this but now i am facing another error 

[TRT] Could not register plugin creator - : :FlattenCorcat_TRT version 1

nvbuf_utils: dmabuf_fd -1 mapped entry not found
nvbuf_utils: Can not get buffer from FD... Exiting...
[gstreamer] gstCamera -- end of stream (E0S)

(python:11425): GStreamer-CRITICAL **:23:24:35.764: gst_mini_object_set_qdatat: assertion 'object !=NULL' failed

I am using raspberry pi camera RV 1.3

dusty-nv, 2021-03-26:
> I am using raspberry pi camera RV 1.3

I don't think rPI camera v1.3 is supported, it is the IMX219-based rPI camera v2 that is supported in JetPack.   You can check if you can view it with `nvgstcapture-1.0`, but I think the v1.3 isn't supported out of the box because the older Omnivision sensor it uses is EOL.

ZeeshanSar, 2021-03-26:
ok now i am using imx219 8-0010 (platform:54080000.vi:4): connected to video

error is 
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink

dusty-nv, 2021-03-26:
Can you view it with `nvgstcapture-1.0`?

ZeeshanSar, 2021-03-26:
yes

dusty-nv, 2021-03-27:
OK, that is good.  Does it work with video-viewer utility from jetson-inference?

ZeeshanSar, 2021-03-27:
Yes it is working as well.

ZeeshanSar, 2021-03-28:
hi dusty how i can use the mix 219 camera? i have an error estremer changed state from ready to paused ==> mysink

dusty-nv, 2021-03-28:
Can you provide the whole terminal output from when you run 'video-viewer csi://0' command?

ZeeshanSar, 2021-03-29:
after command 
nvidia@nvidia-desktop:~ video-viewer csi://0

it opens a window with the video signal
video-viewer: captured 955 frames (1280 x 720)

ZeeshanSar, 2021-03-29:
nvidia@nvidia-desktop:~$ video-viewer csi://0
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1366x768
[OpenGL] glDisplay -- X window resolution:    1366x768
[OpenGL] glDisplay -- display device initialized (1366x768)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1366
  -- height:     768
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvvconv0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter0
[gstreamer] gstreamer message stream-start ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> nvarguscamerasrc0
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
[... 119 lines cut ...]
video-viewer:  captured 64 frames (1280 x 720)
video-viewer:  captured 65 frames (1280 x 720)
video-viewer:  captured 66 frames (1280 x 720)
[OpenGL] glDisplay -- set the window size to 1280x720
video-viewer:  captured 67 frames (1280 x 720)
video-viewer:  captured 68 frames (1280 x 720)
video-viewer:  captured 69 frames (1280 x 720)
[OpenGL] glDisplay -- set the window size to 1280x720
video-viewer:  captured 70 frames (1280 x 720)
video-viewer:  captured 71 frames (1280 x 720)
[OpenGL] glDisplay -- set the window size to 1280x720
video-viewer:  captured 72 frames (1280 x 720)
[OpenGL] glDisplay -- set the window size to 1280x720
video-viewer:  captured 73 frames (1280 x 720)
video-viewer:  captured 74 frames (1280 x 720)
video-viewer:  captured 75 frames (1280 x 720)
video-viewer:  captured 76 frames (1280 x 720)
video-viewer:  captured 77 frames (1280 x 720)
video-viewer:  captured 78 frames (1280 x 720)
video-viewer:  captured 79 frames (1280 x 720)
video-viewer:  captured 80 frames (1280 x 720)
video-viewer:  captured 81 frames (1280 x 720)
video-viewer:  captured 82 frames (1280 x 720)
video-viewer:  captured 83 frames (1280 x 720)
video-viewer:  captured 84 frames (1280 x 720)

closed 2023-04-27 by dusty-nv