dusty-nv/jetson-inference #968: GStreamer not working

rajeshroy402, 2021-03-26:
Hi, I have issues with the remote streaming of my MIPI CSI Camera on my Windows/Linux machine. I have tested the video-viewer on my Jetson Nano 4GB but the RTP is not showing up. 

So, let me take you towards my commands:
I first logged into Jetson nano, then ran the `docker/run.sh` to use this command syntax -- `video-viewer csi://0 rtp://<remote-ip>:1234`

It runs and shows something like this:

```
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
nvbuf_utils: Could not get EGL display connection
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! udpsink host=MSI port=1234 auto-multicast=true
[video]  created gstEncoder from rtp://MSI:1234
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://MSI:1234
     - protocol:  rtp
     - location:  MSI
     - port:      1234
  -- deviceType: ip
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[... 127 lines cut ...]
video-viewer:  captured 56 frames (1280 x 720)
video-viewer:  captured 57 frames (1280 x 720)
video-viewer:  captured 58 frames (1280 x 720)
video-viewer:  captured 59 frames (1280 x 720)
video-viewer:  captured 60 frames (1280 x 720)
video-viewer:  captured 61 frames (1280 x 720)
```

Now, when I use this command  `$ gst-launch-1.0 -v udpsrc port=1234 \
 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! \
 rtph264depay ! decodebin ! videoconvert ! autovideosink` it gives the following result on the terminal 
 
 ```
Setting pipeline to PAUSED ...
nvbuf_utils: Could not get EGL display connection
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
/GstPipeline:pipeline0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
/GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
New clock: GstSystemClock

```

And nothing happens here after... 
Please let me know what's wrong... I am following the same commands on my guest Ubuntu OS and the same issues I have faced.

dusty-nv, 2021-03-26:
Hi @rajeshroy402, can you ping the PC you are sending the RTP stream to from your Jetson?

When you run video-viewer, can you try putting the IP address of your PC here:

``` bash
video-viewer csi://0 rtp://<192.168.1.100>:1234
```

You also may want to try changing the port to 5000 in case there is some firewall on your PC running.  Try disabling firewall if you have one.

Also, on your PC, do you see the RX packets on your network interface going up when the stream is being sent?

rajeshroy402, 2021-03-27:
Hi, thanks for reaching out. 
Tried everything like instructed by you but it's still not working..
I logged into Jetson via USB and then I opened the docker container and ran the rtp command. I have disabled the firewall and also tried the permutation of port values but it is still not helping.
Okay, I will show you my logs for what is happening.. 

```
root@jetson-desktop:/jetson-inference# video-viewer csi://0 rtp://MSI:5000
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device csi://0
[gstreamer] gstCamera pipeline string:
[gstreamer] nvarguscamerasrc sensor-id=0 ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, framerate=30/1, format=(string)NV12 ! nvvidconv flip-method=2 ! video/x-raw ! appsink name=mysink
nvbuf_utils: Could not get EGL display connection
[gstreamer] gstCamera successfully created device csi://0
[video]  created gstCamera from csi://0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: csi://0
     - protocol:  csi
     - location:  0
  -- deviceType: csi
  -- ioType:     input
  -- codec:      raw
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: rotate-180
  -- loop:       0
------------------------------------------------
[gstreamer] gstEncoder -- codec not specified, defaulting to H.264
[gstreamer] gstEncoder -- pipeline launch string:
[gstreamer] appsrc name=mysource is-live=true do-timestamp=true format=3 ! omxh264enc bitrate=4000000 ! video/x-h264 !  rtph264pay config-interval=1 ! udpsink host=MSI port=5000 auto-multicast=true
[video]  created gstEncoder from rtp://MSI:5000
------------------------------------------------
gstEncoder video options:
------------------------------------------------
  -- URI: rtp://MSI:5000
     - protocol:  rtp
     - location:  MSI
     - port:      5000
  -- deviceType: ip
  -- ioType:     output
  -- codec:      h264
  -- width:      0
  -- height:     0
  -- frameRate:  30.000000
  -- bitRate:    4000000
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
------------------------------------------------
[OpenGL] failed to open X11 server connection.
[OpenGL] failed to create X11 Window.
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> nvvconv0
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvvconv0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> nvarguscamerasrc0
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer stream status ENTER ==> src
[... 122 lines cut ...]

rtph264depay : The term 'rtph264depay' is not recognized as the name of a cmdlet, function, script file, or operable program. Check
the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:3 char:2
+  rtph264depay ! decodebin ! videoconvert ! autovideosink
+  ~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (rtph264depay:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

```

And when I try the same thing with my guest OS Ubuntu, it prints this: 
```
rajesh@rajesh-virtual-machine:~/Desktop$ gst-launch-1.0 -v udpsrc port=5000  caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" !  rtph264depay ! decodebin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
/GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```

I have installed this version of GStream on Windows: [GStream](https://gstreamer.freedesktop.org/data/pkg/windows/1.0.7/)

It's not working!

rajeshroy402, 2021-03-30:
Hi, anyone?

h3retek, 2021-04-04:
@rajeshroy402 

Please try this outside of the docker, by compiling jetson-inference. Reason being: I run my Nano headless, and I had some Xauth problems with it, because if you ssh into the Nano with `user` and you run the docker, the docker goes to `root`. Xauth doesn't like this. So, just [compile this project.](https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md)

After this, I got a (delayed [`rtspLatency 2000` when `gstCamera` is created, as seen here:
```
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
```
]) stream to my PC with:
`video-viewer --headless /dev/video0 rtp://10.0.0.2:5000` <- please note the `--headless`

And then on your PC, install `vlc` and create a file on your desktop (or anywhere): `foo.sdp` containing:

```
c=IN IP4 127.0.0.1
m=video 5000 RTP/AVP 96
a=rtpmap:96 H264/90000
```
where 5000 is the port number used with `video-viewer`.

run this file with vlc, and presto: ticktock time

ZhichengSong6, 2021-05-24:
@rajeshroy402 
Have same question and solve by the follow steps.
Try type netstat -ano in cmd and find a local address(I choose 192.168.137.1:139)
 Use this as your remote ip and port.

closed 2023-04-27 by dusty-nv