dusty-nv/jetson-inference #971: Trained model

ZeeshanSar, 2021-03-29:
Hi dusty,

I have trained a model for potatos detection 
When call the detectnet --model=models/potato/ssd-mobilenet.onnx --labels=models/potato/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0

Attached screen pops up.

[image]

dusty-nv, 2021-03-29:
Hi @ZeeshanSar, I'm not able to see the image you uploaded

ZeeshanSar, 2021-03-30:
yeah this is as it is not able to read

ZeeshanSar, 2021-03-30:
[image]

ZeeshanSar, 2021-03-30:
Oh sorry I uploaded it once again

ZeeshanSar, 2021-03-30:
Is it ok for you?

ZeeshanSar, 2021-03-31:
hi dusty, how i can fix this?

dusty-nv, 2021-03-31:
Interesting...that is a lot of bounding boxes :)
It's strange because normally the clustering algorithm would prevent so many bounding boxes.

To rule out any issue with the model itself, can you try running the [`run_ssd_example.py`](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py) script under `jetson-inference/python/training/detection/ssd`?   You would run this on one of your PyTorch .pth checkpoints (not the ONNX model)

ZeeshanSar, 2021-03-31:
it replies with no comand found

dusty-nv, 2021-03-31:
You need to `cd jetson-inference/python/training/detection/ssd` first and look at the `run_ssd_example.py` script for the command-line options

ZeeshanSar, 2021-03-31:
:~/jetson-inference/python/training/detection/ssd$ run_ssd_example.py

dusty-nv, 2021-03-31:
Run with with `python3` command.  You will need to supply arguments to it, like name of image and the .pth model checkpoint to use

ZeeshanSar, 2021-03-31:
can you give me the complete line?

dusty-nv, 2021-03-31:
`python3 run_ssd_example.py mb1-ssd-lite models/<your-model-dir>/<your-checkpoint>.pth <path-to-your-image>`

ZeeshanSar, 2021-04-03:
hi dusty,

is there any way to add pictures of models directly instead of capturing using the camera-capture command?

ZeeshanSar, 2021-04-03:
hi dusty i am having probblem while 

root@nvidia-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --dataset-type=voc --data=data/plant --model=models/plant
2021-04-03 19:32:34 - Using CUDA...
2021-04-03 19:32:34 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/dieased', dataset_type='voc', datasets=['data/plant'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-04-03 19:32:34 - Prepare training datasets.
Traceback (most recent call last):
  File "train_ssd.py", line 214, in <module>
    target_transform=target_transform)
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 33, in __init__
    raise IOError("missing ImageSet file {:s}".format(image_sets_file))
TypeError: unsupported format string passed to PosixPath.__format__
root@nvidia-desktop:/jetson-inference/python/training/detection/ssd# python3 train_ssd.py --dataset-type=voc --data=data/plant --model=models/healthy --batch-size=2 --workers=1 --epochs=1
2021-04-03 19:33:16 - Using CUDA...
2021-04-03 19:33:16 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=2, checkpoint_folder='models/healthy', dataset_type='voc', datasets=['data/plant'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=1, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-04-03 19:33:16 - Prepare training datasets.
Traceback (most recent call last):
  File "train_ssd.py", line 214, in <module>
    target_transform=target_transform)
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 33, in __init__
    raise IOError("missing ImageSet file {:s}".format(image_sets_file))
TypeError: unsupported format string passed to PosixPath.__format__

dusty-nv, 2021-04-05:
> is there any way to add pictures of models directly instead of capturing using the camera-capture command?

Hi @ZeeshanSar, you can use the CVAT tool (CVAT.org) to annotate images, and then export them to Pascal VOC format

dusty-nv, 2021-04-05:
> File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 33, in **init**
> raise IOError("missing ImageSet file {:s}".format(image_sets_file))

Under you dataset in the `ImageSets/Main` folder, you need the following files:

``` bash
$ ls ImageSets/Main
test.txt  train.txt  trainval.txt  val.txt
```

If some of those are missing, copy one of the ones that is there.  For example, if you dont have `test.txt`, then `cp trainval.txt test.txt`

ZeeshanSar, 2021-04-06:
Hi dusty
text.txt  train.txt  trainval.txt  val.txt 

Files are available see attached picture
[image]

dusty-nv, 2021-04-06:
> text.txt, train.txt, trainval.txt, val.txt,

Do you mean `test.txt` is there?

ZeeshanSar, 2021-04-06:
Yes test file is already in the folder

dusty-nv, 2021-04-06:
Are you running inside container or outside?   Are you sure that your path `--data=data/plant` is correct?  

Try running this (inside container if you are using container)

``` bash
cat /jetson-inference/python/training/detection/ssd/data/plant/ImageSets/main/test.txt
cat /jetson-inference/python/training/detection/ssd/data/plant/ImageSets/main/trainval.txt
```

Does it make valid output?

ZeeshanSar, 2021-04-06:
I got an error 
cat /jetson-inference/python/training/detection/ssd/data/plant/ImageSets/main/test.txt: no such file or directory

dusty-nv, 2021-04-06:
OK, so that path doesn't exist.  When you said that "test.txt file is already in the folder", what was the full path of that folder that you found test.txt in?

ZeeshanSar, 2021-04-06:
The path is 

nvidia@nvidia-desktop:~/jetson-inference/python/training/detection/ssd/data/plants/ImageSets/Main

dusty-nv, 2021-04-06:
OK.  Are you running train_ssd.py in the container or outside of container?

If you are using the container, after you start the container, can you:

```
cat python/training/detection/ssd/data/plants/ImageSets/Main/trainval.txt
```

If path not found, please look around for it and find the right path

VeeranjaneyuluToka, 2021-05-05:
@dusty-nv , wondering, do we need all four files (test.txt  train.txt  trainval.txt  val.txt), I am just trying with test.txt and trainval.txt, and am able to train it. would you mind giving the purpose of each file and how they are being used?

dusty-nv, 2021-05-05:
I think you only actually need test.txt and trainval.txt.   trainval.txt is the actual set used for training, test.txt is used to evaluate the model during training at the end of each epoch

closed 2023-04-27 by dusty-nv

daparic, 2023-05-10:
I only sampled 28 JPEGs of chess pieces. I don't see any detection made. In 28 too low? These are regularly formed chess pieces, so no matter how I position the piece they will look the same. Hence, I only have 28 images. Any comments on this?

dusty-nv, 2023-05-10:
@daparic Yes my guess is this is too few, I typically have at least 100 images per object and ideally much more.   Are you doing classification (train.py) or object detection (train_ssd.py) ?   If the later, you can run train_ssd.py with the --validate-mean-ap flag and it will print out the per-class accuracies for you after each training epoch.