dusty-nv/jetson-inference #983: ask something about Coding Your Own Object Detection Program

tianzanaruk, 2021-04-02:
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("csi://0")      # '/dev/video0' for V4L2
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
	

how i can use data while camera finded or detected person 
example detected 1 person GPIO or led ON

dusty-nv, 2021-04-02:
Hi @tianzanaruk, you would loop through each detection and check if it is a person:

``` python
for n in range(net.GetNumClasses()):
   if net.GetClassDesc(n) == 'person':
      person_class_id = n

while display.IsStreaming():
   img = camera.Capture()
   detections = net.Detect(img)

   for detection in detections:
      if detection.ClassID == person_class_id:
         print('detected a person!')
         # do your GPIO here

   display.Render(img)
   display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

tianzanaruk, 2021-04-02:
Thank​ for​ help.​ ill.learn more

tianzanaruk, 2021-04-03:
hi @dusty-nv  i need to merge code for while detected person should start code left in picture

[image]

tianzanaruk, 2021-04-05:
this code 

```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'person':
      		person_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == person_class_id:
         		print('detected a person!')
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```
 and need to merge with this

```
import speech_recognition as sr  
from gtts import gTTS
from playsound import playsound 
from datetime import datetime

r = sr.Recognizer()

with sr.Microphone() as source: 
	playsound("./signal.mp3") 
	audio = r.record(source, duration=5) 
	playsound("./signal.mp3") 

	try:
		text = r.recognize_google(audio, language="th") 
		if "ผม" in text:
			text = text.replace("ผม", "ฉันเองก็")
		if "ครับ" in text:
			text = text.replace("ครับ", "ค่ะ")
		if text == "กี่โมงแล้ว":
			now = datetime.now() 
			text = now.strftime("ขณะนี้เวลา%Hนาฬิกา%Mนาที%Sวินาที")
	except:
		text = "ขอโทษค่ะ"
	tts = gTTS(text, lang="th") 
	tts.save("./answer.mp3") 
	playsound("./answer.mp3")
```

i try to this 

```
import jetson.inference
import jetson.utils
import speech_recognition as sr  
from gtts import gTTS
from playsound import playsound 
from datetime import datetime

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")
r = sr.Recognizer()

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'person':
      		person_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == person_class_id:
         		print('detected a person!')
                       
```
                        with sr.Microphone() as source: 
	                         playsound("./signal.mp3") 
	                        audio = r.record(source, duration=5) 
	                        playsound("./signal.mp3") 

	                        try:
		                         text = r.recognize_google(audio, language="th") 
		                        if "ผม" in text:
		                        	text = text.replace("ผม", "ฉันเองก็")
	                         	if "ครับ" in text:
		                         	text = text.replace("ครับ", "ค่ะ")
		                        if text == "กี่โมงแล้ว":
		                        	now = datetime.now() 
		                          	text = now.strftime("ขณะนี้เวลา%Hนาฬิกา%Mนาที%Sวินาที")
                        	except:
                         		text = "ขอโทษค่ะ"
                        	tts = gTTS(text, lang="th") 
                        	tts.save("./answer.mp3") 
                        	playsound("./answer.mp3")
```
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```

h3retek, 2021-04-05:
> but not work

Please ellaborate...

tianzanaruk, 2021-04-05:
> > but not work
> 
> Please ellaborate...

when i use python3 test.py  
TabError : inconstent yse of tabs and spaces in indentation

h3retek, 2021-04-05:
I welcome you, to the world where `ctr+c` and `ctrl+v` falls apart.

You are facing what is called: formatting.

> The Python “TabError: inconsistent use of tabs and spaces in indentation” error is raised when you try to indent code using both spaces and tabs.

Which text editor are you using?

tianzanaruk, 2021-04-05:
Nothing.txt

dusty-nv, 2021-04-05:
You need to convert the tabs to spaces (or vice versa)

Also, the way your code is structured, it will freeze the video for several seconds while your audio plays/records, but maybe that is ok.  The alternative would be to use Python multiprocessing, and have the microphone running in another thread or process.

tianzanaruk, 2021-04-06:
@dusty-nv freeze​ the​video.it's ok.
But.need to​ know how​ to.structured​ my​ Code.for running

h3retek, 2021-04-06:
Text editor? khrab

Gedit
Nano
Notepad++
...

tianzanaruk, 2021-04-06:
but my code is work? if use text editor?

h3retek, 2021-04-06:
Nope, but you can use text editor to convert all /tabs to /space :)

If you tell me which text editor you use, I can point you to the tutorial on how to do it.

tianzanaruk, 2021-04-06:
ok

tianzanaruk, 2021-04-06:
> Nope, but you can use text editor to convert all /tabs to /space :)
> 
> If you tell me which text editor you use, I can point you to the tutorial on how to do it.

now i'm ready can u help ?

tianzanaruk, 2021-04-06:
Hi @dusty-nv  can't run coz something lost about `with`

[image]

dusty-nv, 2021-04-06:
There still must be leading spaces lurking in your file.  Or if you changed to spaces, there still must be tabs somewhere.  Typically it is around the line of the error...

tianzanaruk, 2021-04-06:
thanks all

closed 2021-04-06 by tianzanaruk

h3retek, 2021-04-07:
@tianzanaruk because life is all about automation... you can check [this post](https://stackoverflow.com/questions/2490686/enforce-spaces-or-tabs-only-in-python-files) as well

have a look at the nifty script: [reindent.py](http://svn.python.org/projects/python/trunk/Tools/scripts/reindent.py)