dusty-nv/jetson-inference #984: Can't.export.onnx

tianzanaruk, 2021-04-02:
[image]

dusty-nv, 2021-04-02:
Hi @tianzanaruk, this can happen if the number of classes in your labels.txt doesn't match the number of classes that your model was trained with.   Does the file `models/test/labels.txt` contain the `BACKGROUND` class?   The `BACKGROUND` should have automatically been added during training.

Can you try using this command line:   `python3 onnx_export.py --model-dir=models/test --labels=models/test/labels.txt`

tianzanaruk, 2021-04-02:
@dusty-nv same error

tianzanaruk, 2021-04-02:
@dusty-nv and​ now​ done.​ insert background​ and​ run​ again​ thank​ for​ help.

dusty-nv, 2021-04-02:
FYI `BACKGROUND` should be the first class in the labels.txt.  train_ssd.py inserts this and then saves a new copy of labels.txt in your model's folder.

tianzanaruk, 2021-04-02:
Hi @dusty-nv But​ my​ detectnet.​ Not​ found​everything

dusty-nv, 2021-04-02:
From your screenshot, it looks like you trained it for 1 epoch.  You'll need to train it for more epochs to get decent result (30 is typical)

Also, try running `detectnet` with `--threshold=0.25` argument to lower the detection threshold (the default threshold is `0.5`)

tianzanaruk, 2021-04-02:
ok thanks

closed 2021-04-02 by tianzanaruk