dusty-nv/jetson-inference #988: download-models.sh

h3retek, 2021-04-04:
Could we add a check to see if the files have already been downloaded and extracted?

That way, if the install gets interrupted for some reason, it does not download the files every time.

dusty-nv, 2021-04-05:
Hi @h3retek, that could be possible, although some users need to re-download models due to corruption, which would complicate matters.  Particularly those in China have problems sometime downloading the files.  

The way I try to do this today, is that the first time you run `cmake ../`, it runs the downloader script, but on future invokations of cmake, it doesn't run the downloader script.  You would need to run the downloader script manually again, at which time you can de-select those models except the new ones you want to download.

The `docker/run.sh` script also does something similar - it only kicks off the downloader script if it hasn't run before or if you have no models at all.

h3retek, 2021-04-06:
> on future invokations of cmake
In my case, cmake never finished :) so, it just kept on downloading the files... on a fast network, no problem, but I "only" have a 10Mb GSM. FeelsBadMan

> re-download models due to corruption
We can check the hash?

dusty-nv, 2021-04-06:
In your case, you may just want to comment out the model downloader from being called in cmake - https://github.com/dusty-nv/jetson-inference/blob/86776674121f071453b64cc5754e628cb6a6d32c/CMakePreBuild.sh#L45

You would then need to run the model downloader manually if/when you wanted to download the models.

While possible, checking the hash would add complications to the script and I would need to keep track of the hashes.

h3retek, 2021-04-06:
Understood.

Thank you for your time.

h3retek, 2021-04-06:
Just one last suggestion:

Maybe modify the `wget` command with `‐‐continue ‐‐timestamping`?

dusty-nv, 2021-04-06:
I wonder if the download was corrupted, would `--continue` keep appending to the corrupted file, or would it restart from the beginning?

If you want to try it out and report back how it works, that would be great.  Thanks!

h3retek, 2021-04-07:
It shall be done sir.

h3retek, 2021-04-08:
I have come across this [article](https://www.fourmilab.ch/documents/corrupted_downloads/)... which is a very interesting read, if you have the time.

[Mr. John Walker](https://www.fourmilab.ch/) created a `perl` script that checks the file's integrity on his `ftp` server. I believe the only secure way to check file integrity is to provide a `md5` or `sha1` check.

Anyway, in the future, I will just follow your advice, skip the downloader in `jetson-inference/CMakePreBuild.sh`... unless of course I am on better internet ^_^

SriniVest, 2021-04-08:
Thanks so much

closed 2023-04-27 by dusty-nv