dusty-nv/jetson-inference #989: Question about Collecting your own Detection Datasets

tianzanaruk, 2021-04-04:
in youtube Jetson AI Fundamentals - S3E5 - Training Object Detection Models (at time 22:43) 

How i can use data while camera finded or detected my collecting.

Example detected 1 John_Deere  for GPIO started and create function led ON

h3retek, 2021-04-05:
[Jetson GPIO](https://github.com/NVIDIA/jetson-gpio)

tianzanaruk, 2021-04-05:
Hi @h3retek  if this code i need to use my collected not ssd-mobilenet-v2

```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'person':
      		person_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == person_class_id:
         		print('detected a person!')
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```

h3retek, 2021-04-05:
> i need to use my collected [images]

1. [Collecting your own Detection Datasets](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md#collecting-your-own-detection-datasets)
2. [Re-training SSD-Mobilenet](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#re-training-ssd-mobilenet)

tianzanaruk, 2021-04-05:
> 1. [Collecting your own Detection Datasets](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md#collecting-your-own-detection-datasets)
> 2. [Re-training [...]

i'm done 2 topic that but i need to use my collected in python code coz my bad and need to learn python

  Ex.  change ssd-mobilenet-v2 to facenet

```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("facenet", threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'face':
      		face_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == face_class_id:
         		print('detected a face!')
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```
but i need my collected 

```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet(`"detectnet --model=models/tractors/ssd-mobilenet.onnx --labels=models/tractors/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes "`, threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'face':
      		face_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == face_class_id:
         		print('detected a face!')
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))
```
can do this?

h3retek, 2021-04-05:
Maichai khrab

Have a look at this [detectnet.py](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py)

Check the argument requirements, this leads me to believe that you can [use this](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md#training-your-model):

```
net = jetson.inference.detectNet("--model=$NET/ssd-mobilenet.onnx --labels=$NET/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            csi://0", threshold=0.5)
```

also run `detectnet.py --help` it will give you a good overview of everything that can be set and sent...

tianzanaruk, 2021-04-05:
ok thanks very much.
i'll try to use 

```
net = jetson.inference.detectNet("--model=$NET/ssd-mobilenet.onnx --labels=$NET/labels.txt \
          --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
            csi://0", threshold=0.5)
```

closed 2023-04-27 by dusty-nv