dusty-nv/jetson-inference #990: RTSP stream

tianzanaruk, 2021-04-04:
i need to insert -- rtp://<remote-ip>:1234 -- in my code

this code

import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("/dev/video0") 
display = jetson.utils.videoOutput("display://0")

for n in range(net.GetNumClasses()):
   	if net.GetClassDesc(n) == 'person':
      		person_class_id = n

while display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	for detection in detections:
      		if detection.ClassID == person_class_id:
         		print('detected a person!')
	display.Render(img)
	display.SetStatus("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

h3retek, 2021-04-05:
Please check out: [MD File Formatting](https://guides.github.com/features/mastering-markdown/)

It will make your posts much more readable for everyone :)

h3retek, 2021-04-05:
Try to insert `rtp://<other pc ip>:1234` over here

```
....
display = jetson.utils.videoOutput("rtp://10.0.0.2:1234")
...
```

Why? Well, since this is your [output stream](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#output-streams).

tianzanaruk, 2021-04-05:
thanks code done.

tianzanaruk, 2021-04-05:
> Try to insert `rtp://<other pc ip>:1234` over here
> 
> ```
> ....
> display = jetson.utils.videoOutput("rtp://10.0.0.2:1234")
> ...
> ```
> 
> Why? Well, since this is your [output stream](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#output-streams).

how i can know pc ip

h3retek, 2021-04-05:
Example, on my network I have:

1. Linux PC
2. Jetson Nano
3. Router

Now:

[Output] 1. On Linux PC, you can run `ifconfig`:
```
~$ ifconfig
...
enp0s31f6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
...
```

there, your inet is the IP of the PC.

[Input] 2. same command on Jetson

3. on the router, if you have admin access to it, you can check the IPs that have been assigned to the devices on your network
[image]

tianzanaruk, 2021-04-05:
support windows PC?

> Example, on my network I have:
> 
> 1. Linux PC
> 2. Jetson Nano
> 3. Router
> 
> Now:
> 
> [Output] 1. On Linux PC, you can run `ifconfig`:
> 
> ```
> ~$ ifconfig
> ...
> enp0s31f6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
>         inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
> ...
> ```
> 
> there, your inet is the IP of the PC.
> 
> [Input] 2. same command on Jetson
> 
> 1. on the router, if you have admin access to it, you can check the IPs that have been assigned to the devices on your network
>    [image]

h3retek, 2021-04-05:
[Google](https://www.google.com/search?q=find+windows+pc+ip&oq=find+windows+pc+ip&aqs=chrome..69i57.7770j0j7&sourceid=chrome&ie=UTF-8)

HoangGit, 2021-09-01:
After checking so many time, I found the solution for this in here https://github.com/dusty-nv/jetson-inference/issues/968

closed 2023-04-27 by dusty-nv