dusty-nv/jetson-inference #993: OpenBLAS warning

skakavyponik, 2021-04-06:
hi,
during train_ssd I get :
OpenBLAS Warning : Detect OpenMP Loop and this application may hang. Please rebuild the library with USE_OPENMP=1 option.
It looks like it shows at each step
I dont have information about steps but only at epoch done 
Thanks for help

dusty-nv, 2021-04-06:
Hmm interesting, I have not seen this warning before.  Are you using the jetson-inference container, or built from source?

skakavyponik, 2021-04-07:
From source, but next time I tried it on docker and it worked without warning 
HW: agx xavier

skakavyponik, 2021-04-07:
ok I am sry 
this one: [https://elinux.org/Jetson_Zoo#PyTorch_.28Caffe2.29](url)  helps a lot 
Thank you and also very good tutorials and trial programs btw

closed 2021-04-07 by skakavyponik