dusty-nv/jetson-inference #997: HOW to convert .onnx file to tensorRT engine file using [jetson-inference]?

leaf918, 2021-04-07:
the engine file may be used in the deepstream applications.

Thanks.

dusty-nv, 2021-04-11:
Hmm, I'm not sure of the particular requirements of the TensorRT engine needed for DeepStream, but you could just run imagenet/detectnet/segnet with your model, and it will create the .engine file for you during the loading process.  That .engine file gets saved to the directory that the original model was located.

Or you could use the trtexec tool (found under `/usr/src/tensorrt/bin`).  You can instruct that tool to save the engine for you from ONNX.

dusty-nv, 2021-05-11:
@priyaganaboor-hash are you trying to run the ONNX model through jetson-inference library?   If so, just copy the ONNX model to Jetson and try to run it through imagenet/detectnet/ect.  It will automatically create TensorRT engine.  The TensorRT engine needs to be created on the type of GPU that you want to run it on, so you can't create TensorRT engine on a PC and run it on a Jetson.   You also can't copy TensorRT engine between Nano and Xavier for example.  But you can copy TensorRT engine between Nano's (because they use same GPU)

If you want to make TensorRT engine for a different purpose, the trtexec tool can save the engine for you.  It is found under `/usr/src/tensorrt/bin` (on Jetson)

leaf918, 2021-05-12:
Hi
I have find the best solution for jetson tensortRT inference scenario.

1.https://github.com/dusty-nv/jetson-inference is a very important and great repo and reference stuff for jetson-like tensorrt runtime.Such like memory control,trt plugin.But the destination of this repo is demonstrating the jetson ability ,especially on detect/classification.But it's not for user special destination,for example,OCR.

2.The best converting tool which can convert pytorch inference model to tensorRT mode(engine file or plan file) is Torch2trt,It will use any onnx model ,it's just use the pytorch model's weights and tensorrt API.any will build the corresponding tensorrt Network object,and it can be save ,build ,then be executed.

3.pytorch -training->torch2trt - save engine file -> deepstream scenario(or jetson-inference repo) .Thanks.

closed 2021-05-12 by leaf918