dusty-nv/jetson-inference #998: CPU vs CUDA time

egamezl, 2021-04-07:
Hi all, a newey here.  I wanted to ask for some help regarding the reported CPU and CUDA processing times. Can some one explain to me what is the difference between them? Thanks in advance.

dusty-nv, 2021-04-07:
Hi @egamezl, the CPU time is the wall-clock time while the CUDA kernels are launched on GPU and the CPU is idle waiting for GPU to finish.  The CUDA time is the time that the GPU actually spend processing the kernels.  In general the CPU time will be slightly more than CUDA time, since there is some lag/overhead to launch the CUDA kernels and wait for them to finish.

There may also be times when the CPU is doing additional processing, for example in object detection models the CPU does the bounding-box clustering.  So the CPU time would be higher in that case.

egamezl, 2021-04-08:
Hi @dusty-nv , thanks a lot for the quick answer. It makes sense to me, specially that CPU time should be slightly higher that CUDA time. 

The question came from running detectnet script and looking at the resulting times, here I am constantly seeing higher CUDA times. Since this program detects objects, it is the case of your last statement, right? Thanks again.

dusty-nv, 2021-04-08:
> The question came from running detectnet script and looking at the resulting times, here I am constantly seeing higher CUDA times.

Some of the kernels don't require CPU synchronization.  For example, during visualization step, CUDA draws the bounding boxes over the image, and the CPU doesn't need to wait for that result because the CPU doesn't directly use the contents of that image.  So the CPU time is low because the CPU just keeps going onto the next step.

However in the network step, CPU needs to wait for GPU after, because CPU needs the results of the bounding boxes so it can cluster then during the post-process stage.   Hope that makes sense.

closed 2023-04-27 by dusty-nv