dusty-nv/jetson-inference #999: Saving the coordinates of bounding boxes of a self-trained model

TheInvincible123, 2021-04-07:
Hi @dusty-nv!
I have trained my own model to detect baseball using Transfer Learning on Jetson Nano by following your tutorials! Thanks a lot for that! Now, I want to save the coordinates of bounding boxes while detecting. Basically, I need them to find the center of the ball. Kindly help me as soon as possible. I will be highly obliged!

dusty-nv, 2021-04-07:
Hi @TheInvincible123, sounds cool - are you using Python?  If so, you can get the coordinates from the detection results that are returned from `net.Detect()`

``` python
for detection in detections:
    cx, cy = detection.Center
```

I will leave it to you how you want to do the saving of these coordinates.

BTW you can find the other detection members here:  https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#detectNet
(look under `jetson.inference.detectNet.Detection` section)

TheInvincible123, 2021-04-07:
Got it! Thank you so much! It really means a lot!

closed 2021-04-07 by TheInvincible123