dusty-nv/jetson-inference #1000: nonfunction treshold during detectnet

skakavyponik, 2021-04-08:
Hi after creating my own dataset and trainning I tried to detect object 
and I still can see results under value 0.9 and bigger than 0.5

Command:
`detectnet --model=models/spider/ssd-mobilenet.onnx --labels=models/spider/labels.txt --treshold=0.9 --input-blob=input_0 --output-cvg=scores --output-bbox=boxes /dev/video0`

dusty-nv, 2021-04-08:
Hi @skakavyponik, there appears typo in your command line:

`--treshold=0.9` should be `--threshold=0.9`

skakavyponik, 2021-04-08:
I am so sorry :D :D its because of my homeland and language :D i am not used to write and pronounce that "h" 
really sry for taking your time on something like this

dusty-nv, 2021-04-08:
No worries, not a problem at all!

closed 2021-04-09 by skakavyponik