dusty-nv/jetson-inference #1001: TensorRT Environment

yqlbu, 2021-04-08:
Hi,

I recently tested the jetson-inference container, and I had the following questions about the TensorRT Environment inside the container:

When I run `pip list`, I did not find `tensorrt` in the list. However, if I run `python3 -c "import tensorrt as trt`, it executes with no error prompt. I wonder if the container already included the `tensorrt` package installed, or more specifically, how does it work inside the container?

[image]

Best,
Kev

dusty-nv, 2021-04-08:
Hi @yqlbu, on Jetson the CUDA/cuDNN/TensorRT headers+libraries are mounted at runtime into the container when it is run with `--runtime nvidia` (which is used in my `docker/run.sh` script).   That is probably why pip is not listing it, but it actually is there.   

For more info, see here:  https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-Container-Runtime-on-Jetson#mount-plugins

yqlbu, 2021-04-08:
> Hi @yqlbu, on Jetson the CUDA/cuDNN/TensorRT headers+libraries are mounted at runtime into the container when it is run with `--runtime nvidia` (which is used in my `docker/run.sh` script). That is [...]

Thanks for the notes. I will try it out!

closed 2021-04-08 by yqlbu