dusty-nv/jetson-inference #1002: Viewing RTP Remotely with GStreamer  - WARNING: erroneous pipeline: no element "udpsrc"

Sutyke, 2021-04-09:
Thanks for All your excellent work!

I run jetson nano CSI camera with latest jetpack 4.5.1.

I would like to ask for an advice on [RTP](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#rtp) streaming with GStreamer. I can stream RTP from jetson to PC with VLC, however latency is very high.high. 

When I run on ubuntu PC:
`gst-launch-1.0 -v udpsrc port=1234 \
 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! \
 rtph264depay ! decodebin ! videoconvert ! autovideosink`

I get :
`WARNING: erroneous pipeline: no element "udpsrc"`

I tried this advice however it didn't work
https://forums.developer.nvidia.com/t/udpsrc-not-recognised-at-gstreamer-in-ubuntu/108973

Any idea how to fix the issue?

dusty-nv, 2021-04-09:
Hi @Sutyke, can you try running `gst-inspect-1.0 udpsrc` on your PC to see if it can find the udpsrc plugin?

Sutyke, 2021-04-10:
@dusty-nv thanks for your quick reply:

I will write solution for 2 problem for others:

1) GStreamer WARNING: erroneous pipeline: no element "udpsrc" 
I haven't had access to this when run `gst-inspect-1.0 udpsrc` I could get access by running command with sudo

2) GStreamer won't show video
I had to remove gstreamer1.0-vaapi by running `sudo apt-get remove gstreamer1.0-vaapi`
See below for more info
https://askubuntu.com/questions/1183367/gstreamer-wont-show-video

closed 2021-04-10 by Sutyke