dusty-nv/jetson-inference #1003: Using Opencv with object detection

AK51, 2021-04-10:
Hi, I can run the objection detection, but I also want to detect the white line and put some text on the same (or another) window.

In the detectnet.py, I try to get the "img:, but fail...

import cv2
...
img. input.Capture()
cv2.imshow("output", img)

I wonder how to get the video input for the cv2 in detectnet.py?

thanks.

dusty-nv, 2021-04-10:
Hi AK51, see this example:  https://github.com/dusty-nv/jetson-utils/blob/master/python/examples/cuda-to-cv.py

AK51, 2021-04-11:
Thanks, but the nano freeze....
I just added a block in imagenet-camera.py, highlighted with "//modified"
I try to create another window by using cv2.imshow(...).
it seems cv2.waitKey(1) cause the freeze. Below is the modified code.
(Or can I highlight the black line in the original output window output.Render(img) )

I just want to make a robot following blackline, do I need to use openCV? Or jetson library has the color function? Or example?

Thanks.

`#!/usr/bin/python3
#
# Copyright (c) 2020, NVIDIA CORPORATION. All rights reserved.
#
# Permission is hereby granted, free of charge, to any person obtaining a
# copy of this software and associated documentation files (the "Software"),
# to deal in the Software without restriction, including without limitation
# the rights to use, copy, modify, merge, publish, distribute, sublicense,
# and/or sell copies of the Software, and to permit persons to whom the
# Software is furnished to do so, subject to the following conditions:
#
# The above copyright notice and this permission notice shall be included in
# all copies or substantial portions of the Software.
#
# THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
# IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
# FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.  IN NO EVENT SHALL
# THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
# LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
# FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
# DEALINGS IN THE SOFTWARE.
#
import cv2
import jetson.inference
import jetson.utils

import argparse
import sys

# parse the command line
parser = argparse.ArgumentParser(description="Classify a live camera stream using an image recognition DNN.", 
                                 formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.imageNet.Usage() +
                                 jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="googlenet", help="pre-trained model to load (see below for options)")
parser.add_argument("--camera", type=str, default="0", help="index of the MIPI CSI camera to use (e.g. CSI camera 0)\nor for VL42 cameras, the /dev/video device to use.\nby default, MIPI CSI camera 0 will be used.")
parser.add_argument("--width", type=int, default=1280, help="desired width of camera stream (default is 1280 pixels)")
parser.add_argument("--height", type=int, default=720, help="desired height of camera stream (default is 720 pixels)")
parser.add_argument('--headless', action='store_true', default=(), help="run without display")

is_headless = ["--headless"] if sys.argv[0].find('console.py') != -1 else [""]

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the recognition network
net = jetson.inference.imageNet(opt.network, sys.argv)

# create video sources & outputs
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv+is_headless)
font = jetson.utils.cudaFont()

# process frames until the user exits
while True:
	# capture the next image
	img = input.Capture()

///////////////////////////////////////////////////
//modified
	rgb_img = img
#print('RGB image: ')
#print(rgb_img)
#bgr_img = jetson.utils.cudaAllocMapped(width=rgb_img.width,height=rgb_img.height,format='bgr8')
	bgr_img = jetson.utils.cudaAllocMapped(width=rgb_img.width,height=rgb_img.height,format='bgr8')
	jetson.utils.cudaConvertColor(rgb_img, bgr_img)
# make sure the GPU is done work before we convert to cv2
	jetson.utils.cudaDeviceSynchronize()

# convert to cv2 image (cv2 images are numpy arrays)
	cv_img = jetson.utils.cudaToNumpy(bgr_img)
#	print('cv_img')
#	print(cv_img)
	cv2.imshow("CV_Output", cv_img)
	if cv2.waitKey(1) & 0xFF == ord('q'):
		break
////////////////////////////////////////////////////////////////////////////

	# classify the image
	class_id, confidence = net.Classify(img)

	# find the object description
	class_desc = net.GetClassDesc(class_id)

	# overlay the result on the image	
	font.OverlayText(img, img.width, img.height, "{:05.2f}% {:s}".format(confidence * 100, class_desc), 5, 5, font.White, font.Gray40)
	
	# render the image
	output.Render(img)

	# update the title bar
	output.SetStatus("{:s} | Network {:.0f} FPS".format(net.GetNetworkName(), net.GetNetworkFPS()))

	# print out performance info
	net.PrintProfilerTimes()

	# exit on input/output EOS
	if not input.IsStreaming() or not output.IsStreaming():
		break

	

`

dusty-nv, 2021-04-11:
I'm not sure about how to use cv2.waitkey() or why it would be freezing, but do you need cv2.waitkey(1) at all since that only waits for 1 millisecond for keypress anyways?   What if you just remove it?

Also, folks generally would use videoOutput interface for display, or OpenCV - but I have not really seen both being used.  Although maybe that original code from imagenet at the bottom is commented out, I can't tell.

For performance reasons, you don't want to call `jetson.utils.cudaAllocMapped()` every frame.  If you only call it once, you only need to call `cudaToNumpy()` once, because it is a persistent mapping (and any changes CUDA make, OpenCV will see).  You can do something like this:

``` python
bgr_img = None
cv_img = None

while True:
   img = input.Capture()
   rgb_img = img

   if bgr_img is None:
      bgr_img = jetson.utils.cudaAllocMapped(width=rgb_img.width,height=rgb_img.height,format='bgr8')
      cv_img = jetson.utils.cudaToNumpy(bgr_img)

   jetson.utils.cudaDeviceSynchronize()

   # proceed with your code
```

closed 2023-04-27 by dusty-nv