dusty-nv/jetson-inference #1005: Custom model training error

Balpoff, 2021-04-11:
Hello, everyone!
I have a problem with training my own model. When I run `detectnet --model=models/ball/ssd-mobilenet.onnx --labels=models/ball/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0`

I get an error: [detectnet error](https://paste.gg/p/anonymous/3f8593a138d948d3bcca3f22fc65755a) 

It is my onnx model: [ssd-mobilenet.onnx](https://drive.google.com/file/d/12Z9H4hS9QYcXurg_2btXqx5hs53k1uXO/view?usp=sharing). I think she’s kind of weird.

Jetpack 4.2 + PyTorch 1.4 + TorchVision 0.5 + tensorrt 6.0.1.10 CUDA Version 10.0.326

dusty-nv, 2021-04-11:
Hi @Balpoff, just wanted to confirm - did you train this with train_ssd.py from the tutorial from this repo?

I'm not exactly sure what the issue may be, however the JetPack/TensorRT version strikes me as a bit older for this model.  I recall adding this support for custom-trained SSD-Mobilenet around the JetPack 4.3/4.4 timeframe.  Which Jetson are you using?   Is it possible for you to re-flash your device with newer JetPack (which will by virtue have newer TensorRT) and try again?

dusty-nv, 2021-04-11:
Looking back through my GitHub docs, yea I believe you need to upgrade to JetPack 4.4 to have a TensorRT version that can properly load the model, because I have this notice on the page:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#setup

> note: first make sure that you have JetPack 4.4 or newer on your Jetson

Balpoff, 2021-04-12:
> did you train this with train_ssd.py from the tutorial from this repo?

Yes

> Which Jetson are you using?

Jetson Nano

> Is it possible for you to re-flash your device with newer JetPack (which will by virtue have newer TensorRT) and try again?

The point why I use raspberry camera v1 (sensor OV5647) right now it works because I found a driver for jetpack 4.2, but as soon as I re-flash jetson nano it will stop working. **Do you know about using it on jetpack 4.4?**

dusty-nv, 2021-04-12:
> The point why I use raspberry camera v1 (sensor OV5647) right now it works because I found a driver for jetpack 4.2, but as soon as I re-flash jetson nano it will stop working. **Do you know about using it on jetpack 4.4?**

Unless you have an updated driver, I don't think so, as OV5647 isn't officially supported in JetPack.  The OV5647 sensor is EOL so we support the newer IMX219-based Raspberry Pi camera v2 in JetPack.

closed 2023-04-27 by dusty-nv