dusty-nv/jetson-inference #1006: detectnet - get the frame name

MauricioLins, 2021-04-11:
Hi @dusty-nv and all,

I have been working with the detectnet.py and everything is working fine, but to close the development I need to store the frame name. I am trying to find where it is writen in the C++ code (videoOutput.cpp and others), but I could't identify it do far.

Could you please just give me a clue how I can get the frame name that is being produced when we have output path?

Thanks a lot.

dusty-nv, 2021-04-12:
Hi @MauricioLins, do you mean where the output image filenames are made (if you are writing a sequence of images)?

That is done here - https://github.com/dusty-nv/jetson-utils/blob/c373f49cf21ad2cae7e4d7da7c41f4fd6473958f/image/imageWriter.cpp#L96

MauricioLins, 2021-04-24:
Thank you @dusty-nv

closed 2021-04-24 by MauricioLins