dusty-nv/jetson-inference #1007: Retraining SSD-Mobilenet for Object Detection

ACRONN, 2021-04-12:
Hi All,

I have been following the “[Re-training SSD-Mobilenet](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md)” tutorial on the repository and the corresponding “[Jetson AI Fundamentals](https://www.youtube.com/watch?v=2XMkPW_sIGg&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=15)” video on YouTube with my Jetson Nano (running JetPack 4.5). I attempted to train a model for 30 epochs built on a dataset of 5500 images (11 different object classes) from Open Images. The entire model creation and training process worked successfully. However, I’m finding that my model struggles to detect any of the trained object classes.

At first, I thought that I did not include enough images in the dataset for each class, or that there were too many object classes in the dataset for the number of images (this could still be the case). However, to test the possibility that my current installation of JetPack and Jetson-Inference (which was built from source) was not working, I installed JetPack and built Jetson-Inference from source on another SD card, and followed the exact steps taken in the "Jetson AI Fundamentals" video to create the retrained “fruit” model (i.e. eight fruit object classes trained for one epoch with 6500 images in the dataset). I then ran this model through the images used in the video. While the model trained in the video was detecting many of the objects in these images at high confidence readings (some achieved greater than 90% confidence, such as the Pineapple, etc.), the identical model I created was struggling to find many of the same objects detected in the video, with those that it did find the confidence was a low %. Although I didn’t think it would make a difference, I then created a new installation of JetPack and Jetson-Inference (this time installing it via the Docker container) on the SD card and followed the same process, with mostly the same results.

Could there be a reason why this is the case? Is there a model/setup (training data, set images, etc.) that is always repeatable so I can try to test my setup for consistency?

I have attached the --debug terminal output log when running the retrained fruit model (Docker container) below.

Any help would be appreciated.

Thank you.

[fruit_debug_log.txt](https://github.com/dusty-nv/jetson-inference/files/6293812/fruit_debug_log.txt)

dusty-nv, 2021-04-12:
Hi @ACRONN, can you delete the .engine file in your model's folder and try running detectnet program again?

Also, you can try the [`run_ssd_example.py`](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py) script which will run a pytorch model checkpoint (.pth file) that you provide the path to, on a test image that you provide.  This will check if the model is working ok on the PyTorch side.

Are you able to run my fruits model ok?   It was trained for 100 epochs.  Here is the link:  https://nvidia.box.com/shared/static/gq0zlf0g2r258g3ldabl9o7vch18cxmi.gz

ACRONN, 2021-04-13:
Hi @dusty-nv, 

Thank you for your reply.

I tried deleting the .engine file and re-running detectnet on my one epoch fruit model. No real change in the results.

I also tried running the "[run_ssd_example.py](https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py)" script on the "[fruit_7.jpg](https://github.com/dusty-nv/jetson-inference/blob/master/data/images/fruit_7.jpg)" image (i.e. an image with a pineapple). Again, poor results (i.e. the program returned zero objects detected).

However, your pre-trained 100-epoch model worked perfectly, with most of the fruit images returning high-confidence detected objects.

Does this mean there is a problem with the PyTorch training process, or were you demonstrating your 100-epoch fruit model when you were running detectnet in the "[Jetson AI Fundamentals](https://www.youtube.com/watch?v=2XMkPW_sIGg&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=15)" video (meaning my problem would come down to the number of epochs I trained my model)?

With respect to the 100-epoch model, did you train this on a Jetson or a more powerful higher end machine?

Thank you.

dusty-nv, 2021-04-13:
Hi @ACRONN, yes the demo in the video was running the 100-epoch model.  It was trained on a Jetson.  I think I would have your model train for more epochs and see if that improves.   To speed things up, you could train it on fewer classes / less data, or collect your own dataset.  It trains quickly on your own datasets because there is typically less images in datasets you collected yourself.

ACRONN, 2021-04-14:
Hi @dusty-nv,

Thank you again for your reply. I will try running my original 11 object classes through 100 epochs to see if that will improve its results over 30 epochs.

Quick question. I am aware that you can specify the `--max-images` flag to limit the number of images that are downloaded from Open Images for specified object classes (using the open images downloader tool). However, is there a way to ensure that each object class being downloaded receives the same/similar number of images (e.g. if I want to download 5500 images in total for 11 object classes for a dataset, how can I ensure each object class receives ~500 images each)?

I attempted to download each object class through the open images downloader tool separately, specifying `--max-images=500` for each (i.e. running the downloader tool a total of 11 times, once for each object class, outputting to the "data" folder for my dataset). However, when I went to train the dataset for one epoch (as a test), it had only trained the model on the most recent object class I had downloaded (i.e. the "eleventh" object class) and thus, the model's "labels.txt" file only listed the most recent object class and the BACKGROUND class. Despite this, on checking the "data" folder for my dataset, I found that the images for all the object classes had downloaded correctly.

Does this mean that each time the downloader tool was run for another object class, any downloaded detection/annotation data for the previous object class was overwritten by the new downloaded detection/annotation data, making the Jetson/PyTorch think the dataset only included one object class? Is there any way of preventing this?

Thank you.

closed 2023-04-27 by dusty-nv