dusty-nv/jetson-inference #1009: Retraining for TFLite

neilyoung, 2021-04-14:
Hi Dusty,

After having achieved pretty good object detection results with Jetson Nano, DeepStream SDK, Resnet and PeopleNet (both 4 classes models), I'm now going to achieve the same performance on another platform (RaspberryPI+Coral TPU). I know it is not your party, but I'm really baffled by the quality of your work, tutorials and documentation and I thought you could provide me some input from your experience, if you like.

Since I was not able to follow successfully the Google tutorial (https://coral.ai/docs/edgetpu/retrain-detection/#using-the-coral-dev-board) for transfer learning on object detection, I thought probably this could work:

1) The aim is to re-train a MobileNet SSD v1 or v2 model for 90 classes to be fit for 4 classes only (person, car, roadsign, bicycle) and run it on a Coral TPU with the same performance I get from a Jetson Nano with DeepStream. Right now I'm achieving  30 fps inference rate at 640x480 for three USB cameras simultaneously on the Nano, 30 fps for each cam. I'm only able to achieve 10 fps per cam with the RPI/Coral tuple. I would like to improve that at least by factor 3. I'm sure, using a class-reduced model would definitely help.

2) So I thought I could follow one of your tutorials first and either re-train an existing model with a sub-dataset or create a new model from scratch with own imagery and annotations. You have provided very nice videos and documentations for that, it shouldn't be too problematic to come to first results with that. That would allow me to run the training on the Jetson GPU and - as far as I hope - I could also use the resulting model together with the DeepStream SDK (?) on the Nano for comparison and tests.

3) Then it would be awesome to make this model "Google TPU ready". As far as I understood you are in a first step converting Torch to ONXX, while I would need a TFLite model from the Torch output, which then would need to be compiled by the edgetpu compiler to an edgetpu.tflite model. I thought I could use maybe this repo in order to make the conversion from Torch to TFLite 

https://github.com/omerferhatt/torch2tflite

I have not started any of these adventures. I was just checking, if this path is promising. May I ask you for your opinion on that?

I hope it is not too blasphemic to ask for Google on a Jetson project.

Kind regards

dusty-nv, 2021-04-14:
Hi @neilyoung, unfortunately having never used the Coral device (and not much hands-on TensorFlow myself), I'm afraid I won't have much experience to lend here.  You could give the torch2tflite converter a shot.  Here is also one that I found for ONNX-to-tflite:  https://github.com/joshuacwnewton/ONNX-to-TFLite

neilyoung, 2021-04-14:
Hi Dusty,

thanks for the quick and honest answer. And for the link. I quickly tested it, it looked promising, but I got stuck https://github.com/joshuacwnewton/ONNX-to-TFLite/issues/1

Anyway, I think I will give it a try and report if successful

neilyoung, 2021-04-14:
Regarding your "fruit" retrain video: I followed it, but the number of images available for exactly the same classes is way less 6000. It is about 1600. Does that make sense?

neilyoung, 2021-04-14:
OK, I was running the training on this set of 1600 images on the Jetson Nano. Besides the fact, that I had to switch to `sudo init 3` in between, the training came through with an avg error of 6%. Note: I did run it with epoch=1 first in order to see, if it passes. The torch -> onxx conversion was fine and also detectnet did run. But it didn't detect anything, means: There was no object annotated with a bounding box and a likelihood in the data/images/test folder. Is this an expected result for epoch = 1 (what I could understand) ?

I'm just asking because I also didn't achieve _any_ detection with the model you are offering for download on the tutorial page (the one with the 100 epochs)

neilyoung, 2021-04-15:
Works

closed 2021-04-15 by neilyoung