dusty-nv/jetson-inference #1010: Training on custom dataset loss is inf or nan

fschiegg, 2021-04-14:
Hi! I'm retraining a ssd model using train_ssd.py. I tried both, the default mobilenet-ssd-v1 and vgg16-ssd networks. On both the Avg loss and Avg Regression loss is infinity or nan for many samples.

I tried the steps proposed in https://github.com/dusty-nv/jetson-inference/issues/814. There I noticed that some samples had a -1 Position. This is caused by voc_dataset, since it subtracts 1 from every coordinate. after I fixed that the loss wasn't nan anymore. Unfortunately it still is inf for most samples.

[image]

I tried different learning rates and batch sizes and freezing the base-net.

dusty-nv, 2021-04-14:
Hi @fschiegg, hmm, normally NaN or inf loss means that something is malformed/corrupt in your annotations - which you have found how to track down.  Assuming that isn't the case with your annotations, since you have already looked through them.  Is your dataset fairly straightforward, or does it have lots of bounding boxes per image, or perhaps overlapping or small bounding boxes?   Without taking a look, I'm not sure how to characterize it.

You may want to start with a "simplified" version of your dataset to see if you can characterize what it is about the training data that is leading to this issue.

fschiegg, 2021-04-14:
Hi @dusty-nv . Thank you very much for your fast response! Indeed, the bounding boxes do often overlap. This didn't cause a problem when I trained a similar network in Tensorflow. Regarding the amount of boxes, usually there are around 2-10 bounding boxes per image (as far as I know thats a normal range).

I'll try to create a simplified case in order to isolate the problem and keep you posted on the issue.

fschiegg, 2021-04-15:
I solved the problem. Indeed, there existed mislabeled samples. Single pixels were accidentally labeled. I guess that those small bounding boxes caused the error, since they have no area after the resizing process. 

Thank you again @dusty-nv  for your help!

closed 2021-04-15 by fschiegg