dusty-nv/jetson-inference #1011: onnx does not appear when i list it in jetson nano and jetson nano does  not identify some cats in the example

Gianni2045, 2021-04-15:
I tried the dusty train the cat dog and it is failing to detect the tiger as a cat it says 51 percent dog please help me ASAP
Also I don't know how to access onnx in jetson nano jetson-inference/python/training/classification to make my own object detection training 
Please tell me how to install onnx so it is visible in this container

Gianni2045, 2021-04-15:
i would appreciate if you can kindly help me get through all these tutorials and get some certificates to be considered one day to work for Nvidia

dusty-nv, 2021-04-15:
You don't need to have `onnx` pip package installed in order to use ONNX.  You should still be able to run `export_onnx.sh` without it and TensorRT doesn't need `onnx` package either.  

How many epochs did you train your cat/dog model for?   The one I have for download on the tutorial page was trained for 100 epochs.

Gianni2045, 2021-04-15:
Thank you very much for your quick response I ran the example for default
amount of epochs 35

Gianni2045, 2021-04-15:
so i trained the example twice 35 epochs in my 4GB nano

Gianni2045, 2021-04-15:
it worked several times training and I dont know why I am now getting OSError: [Errno 12] Cannot allocate memory
I cannot even train again

Gianni2045, 2021-04-15:
please help as soon as possible i am stuck i can't move forward now

dusty-nv, 2021-04-15:
Reboot and try these suggestions to reduce the memory usage of your board:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

Reduce the `--batch-size` (the default is 8) and `--workers` (the default is 2)

Gianni2045, 2021-04-15:
i tried rebooting and i cannot login to GUI environment, it says my nano has bad blocks or corrupted, I do NOT want to reflash my 1TB SSD drive, it is going to take me 1 whole day to get back on track, it says i am logged in tty  even though I am logged in directly
how can I recover my hard drive without reflashing it please??
thank you for your support

dusty-nv, 2021-04-15:
Isn't your Nano running from SD card and you can just reflash that?

Can you use the fsck utility to repair?  https://www.tecmint.com/fsck-repair-file-system-errors-in-linux/

Gianni2045, 2021-04-15:
no it's running from a USB SSD

Gianni2045, 2021-04-15:
thank you for the tip

closed 2023-04-27 by dusty-nv