dusty-nv/jetson-inference #1012: Distribution of Images in Google Open Images Download Using python3 open_images_downloader.py

eanmikale, 2021-04-15:
When the images are downloaded using python3 open_images_downloader.py, is there a way to evenly distribute the number of images in each class, rather than images being heavily clustered in a few classes, with many others only having 1 or 2 images per class? Thank you.

dusty-nv, 2021-04-15:
Hi @eanmikale, you can change how the classes are distributed here:   https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/open_images_downloader.py#L167

The way it is now, when an image limit is imposed, it keeps the same ratio of images-per-class to images in the dataset.

eanmikale, 2021-04-15:
We want the number of images to be divided by the number of classes, to ensure the equal distribution of images. 

max_images = int(args.max_images * (len(images[d]) / total_images))

Could you break down this snippet of code? We are multiplying the max number of images, times the length of the images, divided by the total number of images.

Could we do max_images = int(args.max_images / total_images)) ? Is total_images the total number of images available without memory constraint? Thanks again. 

Regards, 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Wed, 14 Apr 2021 21:50:11 -0500 Dustin Franklin ***@***.***> wrote ----

Hi https://github.com/eanmikale, you can change how the classes are distributed here: https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/open_images_downloader.py#L167

The way it is now, when an image limit is imposed, it keeps the same ratio of images-per-class to images in the dataset.

dusty-nv, 2021-04-16:
> max_images = int(args.max_images * (len(images[d]) / total_images))

This is keeping the same ratios of images per class, just scaled by the max number of images.

I recall balancing the classes before, however some classes would not have enough images for that.   Some classes may only have a few hundred images.  However you can change it as you wish.

eanmikale, 2021-04-16:
Dustin, 

Can't thank you enough. We believe this may maximize the training efficiency and accuracy with limited memory constrains on both the Xavier AGX down to the Nano. Thank you again. 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Thu, 15 Apr 2021 19:54:10 -0500 Dustin Franklin ***@***.***> wrote ----

max_images = int(args.max_images * (len(images[d]) / total_images))

This is keeping the same ratios of images per class, just scaled by the max number of images.

I recall balancing the classes before, however some classes would not have enough images for that.   Some classes may only have a few hundred images.  However you can change it as you wish.

dusty-nv, 2021-04-16:
No problem.  By the way, you can run the pytorch-ssd training code on PC/server with discrete GPU if you have access to one.  You just need to install CUDA+PyTorch and the dependencies on PC.  You can export the ONNX to PC and copy it over to your Jetson device(s)

http://www.infinite8institute.com

---- On Thu, 15 Apr 2021 19:54:10 -0500 Dustin Franklin ***@***.***<mailto:***@***.***>> wrote ----

eanmikale, 2021-04-16:
Makes perfect sense to do it this way. We have a suite of apps we built on the Jetson Xavier and Nano Platforms, but have the bottle neck for commercialization. 

1) How can we easily encapsulate the system once we have it finalized for commercial deployment, for easy replication across other Xavier platforms? 

2) What trouble might we run into creating a Vanilla kernel, attempting to prune much of the unnecessary software on the Linux paltform, to create a leaner system for A.I. deployment? 

Note: The Nano makes it easier with an 8-card SD replicator, but it has less IP and security protection. Thanks!

Regards, 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 08:01:07 -0500 Dustin Franklin ***@***.***> wrote ----

No problem.  By the way, you can run the pytorch-ssd training code on PC/server with discrete GPU if you have access to one.  You just need to install CUDA+PyTorch and the dependencies on PC.  You can export the ONNX to PC and copy it over to your Jetson device(s)

 

 

 

 

 

 

 

 

 

 http://www.infinite8institute.com

dusty-nv, 2021-04-16:
I would look to containers to bundle your application and make it easy to deploy on many Jetson devices.

It's unlikely that you would get an upstream vanilla kernel to run on Jetson without applying many of the patches from the L4T kernel.  Also with the GPU support.   However that is not to say that you couldn't make a leaner OS, while still using the L4T kernel.  For example, here is Yocto port for Jetson:  https://github.com/OE4T/meta-tegra

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 08:01:07 -0500 Dustin Franklin ***@***.***<mailto:***@***.***>> wrote ----

http://www.infinite8institute.com

---- On Thu, 15 Apr 2021 19:54:10 -0500 Dustin Franklin ***@***.***<mailto:***@***.***<mailto:***@***.***%3cmailto:***@***.***>>> wrote ----

eanmikale, 2021-04-16:
Would they have to build it again, or would all the applications, such as our one-click desktop applications run? The containers I've used have only been a terminal, where we are pointed to an ip address for the GUI. We want our apps to run as desktop applications, next to the Google Chrome and Trash bin. Is this possible, without programming each one? 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 13:27:03 -0500 Dustin Franklin ***@***.***> wrote ----

I would look to containers to bundle your application and make it easy to deploy on many Jetson devices.
 
 It's unlikely that you would get an upstream vanilla kernel to run on Jetson without applying many of the patches from the L4T kernel.  Also with the GPU support.   However that is not to say that you couldn't make a leaner OS, while still using the L4T kernel.  For example, here is Yocto port for Jetson: https://github.com/OE4T/meta-tegra

 

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 http://www.infinite8institute.com
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 http://www.infinite8institute.com

dusty-nv, 2021-04-16:
You can run GUI apps inside the container.  The demos in jetson-inference use OpenGL windows inside of container.  Just look at the xhost and device mapping stuff that I do in docker/run.sh script.

For desktop links, I would make a shell script on the desktop that launches the container and the app within it.

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 13:27:03 -0500 Dustin Franklin ***@***.***<mailto:***@***.***>> wrote ----

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 08:01:07 -0500 Dustin Franklin ***@***.***<mailto:***@***.***<mailto:***@***.***%3cmailto:***@***.***>>> wrote ----

http://www.infinite8institute.com

---- On Thu, 15 Apr 2021 19:54:10 -0500 Dustin Franklin ***@***.***<mailto:***@***.***<mailto:***@***.***%3cmailto:***@***.***<mailto:***@***.***%3cmailto:***@***.***%3cmailto:***@***.***%3cmailto:***@***.***>>>> wrote ----

eanmikale, 2021-04-16:
Thank you Dusty. I think we are good here for now. Regards. 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Fri, 16 Apr 2021 13:58:24 -0500 Dustin Franklin ***@***.***> wrote ----

You can run GUI apps inside the container.  The demos in jetson-inference use OpenGL windows inside of container.  Just look at the xhost and device mapping stuff that I do in docker/run.sh script.
 
 For desktop links, I would make a shell script on the desktop that launches the container and the app within it.

 

 
 
 
 
 
 
 
 http://www.infinite8institute.com
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 http://www.infinite8institute.com
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 http://www.infinite8institute.com

closed 2023-04-27 by dusty-nv