dusty-nv/jetson-inference #1013: How to load .engine to mb2-ssd-lite model and run inference on static image, in Python script rather than by commandline

surefyyq, 2021-04-15:
Hi dusty-nv @dusty-nv ! Thanks for your excellent work!

I have successfully  trained a **mb2-ssd-lite** model on my custom dataset and completed the **Re-training SSD-Mobilenet**.
After running the commandline:  

> `detectnet --model=models/fruit/ssd-mobilenet.onnx --labels=models/<dataset>/labels.txt \
>           --input-blob=input_0 --output-cvg=scores --output-bbox=boxes \
>             "<path to image>.jpg" <path to save image>.jpg`

 I have an .onnx and .engine file, as excepted.

**Now how should I do to instantiate a mb2-ssd-lite **detectNet** and load my .engine file to it within a pyhton script rather than by commandline?**

I'm a little confused about the `net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)` in python/examples/detectnet.py
Caption says the parameter includs _network(string), argv(strings) and threshold(float)_. But in argv there also is  arguments  _--model=MODEL_ and _--threshold=THRESHOLD_

Question is :
1. Does the jetson.inference.detectNet can only create one of the built-in model, but mb2-ssd-lite not included?
2. If I want to run the model by tensorRT, do I have to load the .onnx or .engine into the model, and how?
3. Can you provide a python example about detectNet processing on single static image?

Thank U!

dusty-nv, 2021-04-15:
Hi @surefyyq, you can create it like this in Python:

``` python
net = jetson.inference.detectNet(arvg=['--model=models/fruit/ssd-mobilenet.onnx', '--labels=models/fruit/labels.txt' , '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes', '--threshold=0.5'])
```

surefyyq, 2021-04-20:
Hi @dusty-nv 
it works perfectly for me and I'm closing this issue, thank u!

closed 2021-04-20 by surefyyq

VeeranjaneyuluToka, 2021-05-04:
Hi @dusty-nv , the api jetson.inference.detectNet(...) takes .onnx as input and creates trt engine if i am correct. wondering can not we use the engine which is created already when we run first time and run the inference for feature samples?

dusty-nv, 2021-05-04:
Hi @VeeranjaneyuluToka, it automatically looks for the .engine file when you try to load the .onnx.  So continue to just pass the .onnx filename.  The first time you load it, it will take longer to make the engine (and then it saves it to disk).  The second time you load it, it should load much faster.

VeeranjaneyuluToka, 2021-05-04:
Hi @dusty-nv , thanks for quick reply. Would you mind sharing any sample which i can refer to run the inference on my images. Is it something that i can refer (python/examples/my-detection.py) and modify so that it will take my images rather than videosource provided in sample?. And what is the best way to understand the complete source code in this github?

dusty-nv, 2021-05-04:
videoSource can run on images and directories of images, see here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#image-files

sas9624, 2021-08-11:
Hello @surefyyq 

I have downloaded the mb2-ssd-lite as suggested from upstream repo https://storage.googleapis.com/models-hao/mb2-ssd-lite-mp-0_686.pth and trained the fruit dataset. However I couldn't convert it to ONNX model. It gave me the errors saying about 

> 	size mismatch for classification_headers.5.weight: copying a param with shape torch.Size([54, 64, 1, 1]) from checkpoint, the shape in current model is torch.Size([54, 256, 3, 3]).
> 	size mismatch for regression_headers.5.weight: copying a param with shape torch.Size([24, 64, 1, 1]) from checkpoint, the shape in current model is torch.Size([24, 256, 3, 3]).

May I ask some questions? 
1. How did you convert mb2-ssd-lite to ONNX format? Did you use `python3 onnx_export.py --model-dir=models/fruit` as suggested in the repo?
2. Where did you download the pre-trained model before you trained your own dataset? 

Thank you very much