dusty-nv/jetson-inference #1014: How to run a retrained SSD model with the DeepStream SDK?

neilyoung, 2021-04-15:
Hi Dusty,

would it be possible to have a look onto this thread? https://forums.developer.nvidia.com/t/onnx-model-on-deepstream5-0-nvinfer-error-could-not-find-nms-layer-buffer-while-parsing/163580/16

Status: I was able to successfully retrain the SSD-MobileNetv1 model following your fruit tutorial. I was able to run it with the DeepStream 5.1 SDK. I'm currently not able to derive the correct `classId` from the results. Also the `for` loop in the bbox parser lib seems to be incorrectly terminated, which leads to a crash.

However, most of the displayed bounding boxes make sense somehow, even though the displayed class is all the time "BACKGROUND".

neilyoung, 2021-04-16:
@dusty-nv I retrained the model with apple, banana and oranges only. THE SAME RESULTS. I don't get it.

But what is even worse: My bbox parser is not working anymore. It shows bananas everywhere :))

The training this morning was very different from yesterday. While yesterday with the whole 8 fruit set I was only able to fetch 1600 image, today with the 3 fruits I fetched > 3000. Does that make sense? Also - yesterday the training took about 2 hours max, today about 6. With less classes. Does that all make sense?

But more importantly: No improvement and detections everywhere. 

I have published the bbox code here. I'm not sure, if it is bug free. Would you mind to cross check? Also - is there any means to prove, if the assumptions regarding the buffer layout are met by the input? 

https://github.com/neilyoung/nvdsinfer_custom_impl_onnx

neilyoung, 2021-04-16:
OK, the detection is fine now and my bbox parser is also ok. What remains is the poor performance :(

dusty-nv, 2021-04-16:
If you require higher performance than the PyTorch/ONNX-based SSD-Mobilenet-v2 can give you, then I would start looking to Transfer Learning Toolkit, which can prune models that it trains for higher performance.  That is how those DeepStream-based object detection models have higher performance.  There are a bunch of pre-trained detection models from TLT that you can try (like tlt-peoplenet for example)

neilyoung, 2021-04-16:
Well, yes. You mentioned that. I'm currently using v1, would v2 perform better? TLT is a good hint, but tbh I was just trying to double check if this reduction gives a boost. I didn't expect it to be that lame. And the primary target is to create a class reduced model for the RPI/Coral.

dusty-nv, 2021-04-16:
From your trtexec perf results from the [forum thread](https://forums.developer.nvidia.com/t/onnx-model-with-jetson-inference-using-gpu/142938/38?u=dusty_nv), it seems 2x faster than the full model (44 FPS vs ~22.5FPS).  And regarding the batching, if your GPU is already fully utilized at 100% from batch_size=1, then it can be difficult for higher batch sizes to substantially improve the performance.

neilyoung, 2021-04-16:
Could you please point me to both result entries? I'm having pretty much the same performance with cameras. Already yesterday the total inference fps was 3x15 = 45 (3 cams) or 2x22.5 = 45 (2cams). Only with one camera it was 30 fps, just because the input was 30 fps. I don't see factor 2 tbh.

neilyoung, 2021-04-16:
I also believe I have posted an outdated trtexec result. This was taken to a time, when all was messed up somehow. Will update

dusty-nv, 2021-04-16:
I am referencing the single-stream inferencing times, where the 90-class SSD-Mobilenet on Nano does ~22.5 FPS

neilyoung, 2021-04-16:
Ah, ok. Yes, that makes sense. Didn't get it.

neilyoung, 2021-04-17:
@dusty-nv I restarted a training for one class "Person" only, limiting the images to 5000, epoch 30, workers 1, batch-size 2. It run for 10 hours, then it stopped suddenly.  I restarted with "--resume", but it looked insane after a while. I restarted from scratch, but soon this was shown on the screen:

[image]

That doesn't look sane to me. It looks like the Nano Dev Kit is a bit overwhelmed by this.

What do you think? Would your "jetson-inference" project for the training also work on an ordinary Linux laptop with GPU or in a Google Colab?

neilyoung, 2021-04-17:
@dusty-nv I was trying to download an original MobileNet SSD v1 ONNX model to no avail so far. I would like to check the performance win achieved by the re-training. What I got was an UINT8 model, which didn't run on the Nano. Do you have a pointer, where to download a matching model? Or should I retrain the *.pth with epoch=1 and export to ONNX?

dusty-nv, 2021-04-17:
Yes you can run pytorch-ssd on PC, GPU is recommended.

There is a fruits model on my page that is SSD-Mobilenet-v1 ONNX trained for 100 epochs.

neilyoung, 2021-04-17:
Hmm. Yes, I know, I downloaded that. I wanted to have the original ONNX, with 90 objects. I already tested yours, exactly the same performance as reported.

dusty-nv, 2021-04-17:
That is a Tensorflow UFF model, not ONNX.
There is no 'original' SSD-Mobilenet PyTorch ONNX model that is used.  The mobilenet backbone was pretrained on ImageNet I believe.

neilyoung, 2021-04-17:
OK, back to this:

> I am referencing the single-stream inferencing times, where the 90-class SSD-Mobilenet on Nano does ~22.5 FPS

How did you come to this result? What did you use? What model file and what app?

And maybe you could also comment the picture above. It the Nano too weak?

dusty-nv, 2021-04-17:
That is the UFF 90-class SSD-Mobilenet model.  Through detectnet program it gets 22.5 FPS, sometimes up to 25 FPS.  Although that isn't an exact apples-to-apples comparison with the ONNX models because the PyTorch-based DNN architecture may be slightly different.  However the ONNX is a lot easier to export.

neilyoung, 2021-04-26:
@dusty-nv I was able to pass the entire chain: Retrain with pytorch following your tutorial, convert to ONNX, convert ONNX to TFLite and try to compile the TFLite to edgetpu.tflite. The last step failed with "Model not quantized". I have provided a Colab for the entire steps and described my approach here https://github.com/google-coral/edgetpu/issues/367.

The Colab can be found here https://colab.research.google.com/drive/1y5YlBFzMAJmKR-qRnT7jzgzGRPtJ6AbM#scrollTo=oxXtqPGRrm9Z. 

Now my question: The error "Model not quantized": Could that be caused by the fact, that the base model was not trained "quantization aware" (whatever this means https://pytorch.org/docs/stable/quantization.html)? If so, would you know a way to change that?

dusty-nv, 2021-04-26:
Since TensorRT can do INT8 calibration while building the engine, I haven't needed to train with quantization before and am not sure how to properly do that.  It appears that PyTorch can quantize an existing model though:

* https://pytorch.org/docs/stable/quantization.html

Or there is a tool for it in onnxruntime:

* https://www.onnxruntime.ai/docs/how-to/quantization.html

neilyoung, 2021-04-26:
Yes, thanks. I have seen at least the first reference also.  I wouldn't know, what the "model" is to be used as input for this operation.

But your second link is promising. Maybe I can post-process the resulting ONNX model that way. Will try

neilyoung, 2021-04-26:
Hmm, no seems to not work. The creation of a quantised model from the original seems to produce something, but the attempt to validate after reload fails:

onnx_model = onnx.load('/content/pytorch-ssd/models/person/ssd-mobilenet.quant.onnx')
onnx.checker.check_model(onnx_model) 

```
ValidationError                           Traceback (most recent call last)
<ipython-input-14-8f48910da94f> in <module>()
      7 # Convert model.onnx to Tensorflow
      8 onnx_model = onnx.load('/content/pytorch-ssd/models/person/ssd-mobilenet.quant.onnx')
----> 9 onnx.checker.check_model(onnx_model)
     10 tf_rep = prepare(onnx_model)
     11 tf_rep.export_graph('ssd-mobilenet.quant')

/usr/local/lib/python3.7/dist-packages/onnx/checker.py in check_model(model, full_check)
    102         if sys.getsizeof(protobuf_string) > MAXIMUM_PROTOBUF:
    103             raise ValueError('This protobuf of onnx model is too large (>2GB). Call check_model with model path instead.')
--> 104         C.check_model(protobuf_string)
    105         m = model
    106     if full_check:

ValidationError: Node (Slice_176) has input size 1 not in range [min=3, max=5].

==> Context: Bad node spec for node. Name: Slice_176 OpType: Slice
```

dusty-nv, 2021-04-26:
> I wouldn't know, what the "model" is to be used as input for this operation.

This would be one of your PyTorch checkpoints (the .pth file with the lowest loss).   You would need to load your checkpoint like this script does:  https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/run_ssd_example.py#L34

neilyoung, 2021-04-26:
Ah, yes. Let's try this...

neilyoung, 2021-04-26:
Stupidity continues....

https://github.com/google-coral/edgetpu/issues/367#issuecomment-826922129

neilyoung, 2021-04-26:
But - as always with Google - one step back and it works...

`!edgetpu_compiler -s 'ssd-mobilenet.tflite'  -m 10
`

I don't get that

closed 2023-04-27 by dusty-nv