dusty-nv/jetson-inference #1017: cudaToNumpy Error

assulthoni, 2021-04-19:
Hello, im facing this issue. any ideas?

```bash
Traceback (most recent call last):
  File "test_converter.py", line 35, in <module>
    cv_img = jetson.utils.cudaToNumpy(bgr_img)
AttributeError: module 'jetson.utils' has no attribute 'cudaToNumpy'
```

I also get this warning 

```bash
jetson.utils -- compiled without NumPy array conversion support (warning)
jetson.utils -- if you wish to have support for converting NumPy arrays,
jetson.utils -- first run 'sudo apt-get install python-numpy python3-numpy'
```

i did install python-numpy but it doesnt work

dusty-nv, 2021-04-19:
You would need to re-build/re-install after installing numpy:

``` bash
$ sudo apt-get install python-numpy python3-numpy
$ cd /path/to/your/jetson-inference
$ cd build
$ cmake ../
$ make
$ sudo make install
```

assulthoni, 2021-05-01:
thanks @dusty-nv it's work!

closed 2021-05-01 by assulthoni

JuanMReyesT, 2023-12-06:
Hi, I'm having the same issue with my jetson nano, but reinstalling and rebuilding after installing numpy seems to have the same result, if you kindly could explain me how did you manege to make it work I will be very greatfull it you, I hope someone could read this, thank you in advance 

best regards 
    
JuanM