dusty-nv/jetson-inference #1018: Install "tkinter' python package in docker

Parham-khj, 2021-04-19:
Hi there,

My python project requires **'tkinter'** and 'matplotlib' packages. Although I could successfully install 'matplotlib' inside docker (by pip3 install matplotlib --verbose), I haven't found a way to install  'tkinter' package. I was wondering if you could help me out with your thoughts. 

The exact error from python compiler running in docker is:

Traceback (most recent call last):
  File "agx_meat_identification.py", line 9, in <module>
    from sort import *
  File "/AGX_deployment/sort.py", line 24, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.6/dist-packages/matplotlib/pyplot.py", line 2336, in <module>
    switch_backend(rcParams["backend"])
  File "/usr/local/lib/python3.6/dist-packages/matplotlib/pyplot.py", line 276, in switch_backend
    class backend_mod(matplotlib.backend_bases._Backend):
  File "/usr/local/lib/python3.6/dist-packages/matplotlib/pyplot.py", line 277, in backend_mod
    locals().update(vars(importlib.import_module(backend_name)))
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.6/dist-packages/matplotlib/backends/backend_tkagg.py", line 1, in <module>
    from . import _backend_tk
  File "/usr/local/lib/python3.6/dist-packages/matplotlib/backends/_backend_tk.py", line 6, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'

dusty-nv, 2021-04-19:
Hi, I do this via apt:  apt-get install python3-tk

Parham-khj, 2021-04-20:
thanks @dusty-nv 
Have you got any solution for "scikit-image' installation in docker?

dusty-nv, 2021-04-20:
> Have you got any solution for "scikit-image' installation in docker?
Hmm not sure - although the [l4t-ml container](https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ml) installs scikit-learn.  I'm not sure if that installs scikit-image too.

closed 2023-04-27 by dusty-nv