dusty-nv/jetson-inference #1020: Detection Outputs to SQL

DJT777, 2021-04-23:
Dusty,

I want to create an SQL entry for every time the detect net model makes a detection using your 10 lines of code example. Is this possible to do using python, and could you give me some insight on where the outputs of the model can be accessed?

Thank you

dusty-nv, 2021-04-23:
Hi @DJT777 , sure - the detections are output into a list of `detectNet.Detection` objects:

``` python
detections = net.Detect(img)

for detection in detections:
   print(detection.ClassID)
   print(detection.Confidence)
   print(detection.Left)
   print(detection.Top)
   # ect
```

Here are the members:

```
Detection = <type 'jetson.inference.detectNet.Detection'>
Object Detection Result
 
----------------------------------------------------------------------
Data descriptors defined here:
 
Area
    Area of bounding box
 
Bottom
    Bottom bounding box coordinate
 
Center
    Center (x,y) coordinate of bounding box
 
ClassID
    Class index of the detected object
 
Confidence
    Confidence value of the detected object
 
Height
    Height of bounding box
 
Instance
    Instance index of the detected object
 
Left
    Left bounding box coordinate
 
Right
    Right bounding box coordinate
 
Top
    Top bounding box coordinate
 
Width
     Width of bounding box
```

For SQL, I have used SQLite before (through sqlite3 in python) on Jetson before without issue.

DJT777, 2021-04-23:
@dusty-nv 

You're awesome. If I have any questions I will use this thread.

DJT777, 2021-04-23:
@dusty-nv
Is there a way to display on screen detection of only one class? Like filter out detections of other things based on class ID. It seems like the code you offered allows choosing detections based on class ID, for example, for output to SQL. I am wondering if something similar can be done for displaying detection.

I.e. allow only dogs to be displayed on screen.

closed 2023-04-27 by dusty-nv