dusty-nv/jetson-inference #1021: Own trained detection model's errors in "detectnet"

WhoseAI, 2021-04-24:
Hi Dusty,

I trained my detection models by using Open Images dataset, everything is OK but the last step get some error,

Device： NVIDIA AGX Xavier
Jetpack：4.5.0
Pytorch version： 1.6.0

The commands are:
1. Downloads dataset from Open Images
$ python3 open_images_downloader.py --class-names="Computer Monitor, Tablet computer, mobile phone, bottle, laptop" --data=data/office --max-images=2000
2. Train model:
$ python3 train_ssd.py --data=data/office --model-dir=models/office
3. Convert to ONNX
$ python3 onnx_export.py --model-dir=models/office

Everything are OK in 1/2/3
But the last step occure some errors:
4.  Detectnet inference
$ detectnet  --model=models/office/ssd-mobilenet.onnx --labels=models/office/labels.txt --input-blob=input_0 --output-cvg=scores --output-blob=boxes /dev/video1 rtp://10.11.11.11:1234

**the error messages are:
[TRT]    ../rtSafe/cuda/genericReformat.cu (1294) - Cuda Error in executeMemcpy: 1 (invalid argument)
[TRT]    FAILED_EXECUTION: std::exception
[TRT]    failed to execute TensorRT context on device GPU
[cuda]      invalid resource handle (error 400) (hex 0x190)
[cuda]      /home/nvidia/nvme120/dusty-nv/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:685**

BTW, this environment can run "detecetnet /dev/video1 rtp://10.11.11.11:1234" correctly.

Best regrads
Hawk Chen

WhoseAI, 2021-04-24:
Hi Dusty,

I found another error during onnx_export.py as below,
**[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1**
Now my TensorRT version is 7.1.3

Is it correlative?

Hawk

dusty-nv, 2021-04-26:
> detectnet --model=models/office/ssd-mobilenet.onnx --labels=models/office/labels.txt --input-blob=input_0 --output-cvg=scores --output-blob=boxes /dev/video1 rtp://10.11.11.11:1234

Can you try running your custom model on just a single test image first to see if it runs properly?

> [TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1

You can ignore this warning, it is not needed.

WhoseAI, 2021-04-27:
Hi Dustin,

I found the major issue may come from the following messages:
----------------------------------------------------------------------------------
[TRT]    device GPU, models/mb1-ssd.onnx initialized.
**[TRT]    detectNet -- number object classes:  4**
[TRT]    detectNet -- maximum bounding boxes:  3000
[TRT]    detectNet -- loaded 8 class info entries
**[TRT]    detectNet -- didn't load expected number of class descriptions  (8 of 4)**
[TRT]    detectNet -- number of object classes:  4
----------------------------------------------------------------------------------
I tried downloading from OpenImages in 4~9 classes, and follow all the steps, but final always show "**number object classes:4** " in "detectnet" step.

BTW, Is there any problem on the classes names with SPACE ? The labels.txt is as below,
**BACKGROUND
Bottle
Computer monitor
Human hand
Laptop
Mobile phone
Tablet computer
Tool
**

Best regards
Hawk

WhoseAI, 2021-04-27:
Can you try running your custom model on just a single test image first to see if it runs properly?

I tried several times, with the same issue of  **[TRT] detectNet -- didn't load expected number of class descriptions (8 of 4)**

But I try the "FRUIT" dataset, run correctly. My OFFICE dataset with the same step......
Thanks

dusty-nv, 2021-04-27:
> **[TRT] detectNet -- didn't load expected number of class descriptions (8 of 4)**

It means that your model actually has less classes than your label file does.

Can you try deleting the .engine file in your model's directory - maybe it is running off of old version of model.

closed 2023-04-27 by dusty-nv