dusty-nv/jetson-inference #1022: Run my-detection.py or detectnet.py with video instead of camera

DJT777, 2021-04-25:
Dusty,

I am looking through the code and I can see that there is an option to run video files through the detectnet and my-detection.

I am wondering if you can show me how it'd be done on the command line and also through changing the code for my-detection.

DJT777, 2021-04-25:
Using this code gave me the camera feed:

```
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource("/media/user/SD/videos/cardinals.mp4")      # '/dev/video0' for V4L2
display = jetson.utils.videoOutput("display://0") # 'my_video.mp4' for file

```

DJT777, 2021-04-25:
attempting to use detectnet.py

`detectnet.py input_URI=file:///media/user/SD/jetson-inference/python/examples/indiana-birds.mp4`
```

[video]  videoSource -- unsupported protocol (input_uri=file)
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 54, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
```
even though input_URI appears to be an argument in detectnet.py

`parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")`

dusty-nv, 2021-04-25:
Here's the page that talks about the different video I/O, including video files:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md

closed 2023-04-27 by dusty-nv