dusty-nv/jetson-inference #1023: Pytorch does not install building from source does not install

Gianni2045, 2021-04-25:
I have tried many times to install pytorch and when i check if it is installed
it says illegal operation in python shell core dumped

I follow all the instructions in setting up the nano this is extremely frustrating and the github pages and nvidia help section does not help

Gianni2045, 2021-04-25:
Can you please help me get this thing going on my ssd I USB in which i am able to boot my jetpack 4.5?

dusty-nv, 2021-04-26:
Hi @Gianni2045, please refer to this post:

https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-8-0-now-available/72048/769?u=dusty_nv

My guess is you had previously upgraded your version of pip/pip3, and there is a bug in numpy 1.19.5 that only gets installed if you had upgraded pip.  Try running `export OPENBLAS_CORETYPE=ARMV8` beforehand or downgrading to numpy 1.18.2

closed 2023-04-27 by dusty-nv