dusty-nv/jetson-inference #1024: Error When Running open_images_downloader.py

DJT777, 2021-04-25:
Hey dusty, I am trying to work through the transfer learning portion of your tutorial. I am getting some errors when attempting to use the open_images_downloader.py. I am wondering if you could have a look and help figure out why I am getting these errors.

```
user@tx1:/media/user/SD/jetson-inference/python/training/detection/ssd$  python3 open_images_downloader.py --class-names "Apple,Orange,Banana,Strawberry,Grape,Pear,Pineapple,Watermelon" --data=data/fruit
2021-04-25 16:53:50 - Requested 8 classes, found 8 classes
2021-04-25 16:53:50 - Read annotation file data/fruit/train-annotations-bbox.csv
^CTraceback (most recent call last):
  File "open_images_downloader.py", line 144, in <module>
    annotation_set = pd.read_csv(annotation_file)
  File "/usr/lib/python3/dist-packages/pandas/io/parsers.py", line 709, in parser_f
    return _read(filepath_or_buffer, kwds)
  File "/usr/lib/python3/dist-packages/pandas/io/parsers.py", line 455, in _read
    data = parser.read(nrows)
  File "/usr/lib/python3/dist-packages/pandas/io/parsers.py", line 1069, in read
    ret = self._engine.read(nrows)
  File "/usr/lib/python3/dist-packages/pandas/io/parsers.py", line 1839, in read
    data = self._reader.read(nrows)
  File "pandas/_libs/parsers.pyx", line 902, in pandas._libs.parsers.TextReader.read
  File "pandas/_libs/parsers.pyx", line 924, in pandas._libs.parsers.TextReader._read_low_memory
  File "pandas/_libs/parsers.pyx", line 1001, in pandas._libs.parsers.TextReader._read_rows
  File "pandas/_libs/parsers.pyx", line 1130, in pandas._libs.parsers.TextReader._convert_column_data
  File "pandas/_libs/parsers.pyx", line 1175, in pandas._libs.parsers.TextReader._convert_tokens
  File "pandas/_libs/parsers.pyx", line 1221, in pandas._libs.parsers.TextReader._convert_with_dtype
  File "/usr/lib/python3/dist-packages/pandas/core/dtypes/common.py", line 777, in is_integer_dtype
    def is_integer_dtype(arr_or_dtype):
KeyboardInterrupt
user@tx1:/media/user/SD/jetson-inference/python/training/detection/ssd$  python3 open_images_downloader.py --class-names "Apple" --data=data/fruit
2021-04-25 16:54:50 - Requested 1 classes, found 1 classes
2021-04-25 16:54:50 - Read annotation file data/fruit/train-annotations-bbox.csv
Killed

```

DJT777, 2021-04-26:
I was able to solve this by cloning the repository onto my laptop then downloading and transferring the data folder to the jetson.

dusty-nv, 2021-04-26:
> I was able to solve this by cloning the repository onto my laptop then running the download script and transferring the data folder to the jetson.

OK gotcha - the `Killed` message means that it ran out of memory.   To free up additional memory, you may want to try these suggestions:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

closed 2023-04-27 by dusty-nv