dusty-nv/jetson-inference #1025: I am not getting the accuracy on classifying cat dog images as the example

Gianni2045, 2021-04-25:
the example classifies the tiger as 94% confidence that is a cat, I am only getting %85 that is a cat percent and I ran the epochs for the default of 34 epochs
for this image $DATASET/test/cat/01.jpg cat.jpg
can you please help me asap because I am dying to know the answer

Gianni2045, 2021-04-26:
please dusty be kind and respond as soon as you  are able to
I am running this on my ssd usb 1 terabyte hard drive

dusty-nv, 2021-04-26:
I trained my cat_dog model for 100 epochs, so that may be a difference.  My model that you can try is here:

https://nvidia.box.com/s/zlvb4y43djygotpjn6azjhwu0r3j0yxc

BTW, if you re-train your model and export it to ONNX again, remember to delete the .engine file in your model's directory.  Then TensorRT will re-generate the engine for your new model.  Otherwise TensorRT will run your old cached model.

Gianni2045, 2021-04-26:
thank you very much

Gianni2045, 2021-04-27:
i deleted the .engine file and retrained the model again for 65 epochs so it resumed at the 34 epochs to complete 100 and get your results and it still did not work any thoughts please?

Gianni2045, 2021-04-27:
it's still not working, am i deleting the file correctly?? can you please help??

dusty-nv, 2021-04-27:
If you see TensorRT running awhile when you ran imagenet/imagenet.py, then yes you deleted it correctly.

I would not worry if you get a different confidence value, it may down to the random initialization of the weights of your model vs mine (when you make a new model, the weights are randomly initialized).   I believe your training is still working OK, and you can move on to collecting your own data if you wish.

Gianni2045, 2021-04-27:
ok thank you very much I understand

closed 2023-04-27 by dusty-nv