dusty-nv/jetson-inference #1026: 'NoneType' object has no attribute shape - when running train_ssd.py

DJT777, 2021-04-25:
Hey Dusty,

I am trying to retrain using your tutorial for retraining mobilenet, but with a single class for birds instead. I used this command on my laptop then transferred over the data

`python3 open_images_downloader.py --max-images=5000 --class-names "Bird" --data=/data/bird`

That produced a dataset, but when I went to train I got this error. I am wondering if you will help me figure this one out.

```
user@tx1:/media/user/SD/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --data=data/bird --model-dir=models/bird --batch-size=4 --epochs=30
2021-04-25 17:30:24 - Using CUDA...
2021-04-25 17:30:24 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/bird', dataset_type='open_images', datasets=['data/bird'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-04-25 17:30:24 - Prepare training datasets.
2021-04-25 17:30:24 - loading annotations from: data/bird/sub-train-annotations-bbox.csv
2021-04-25 17:30:24 - annotations loaded from:  data/bird/sub-train-annotations-bbox.csv
num images:  4338
2021-04-25 17:30:38 - Dataset Summary:Number of Images: 4338
Minimum Number of Images for a Class: -1
Label Distribution:
	Bird: 11700
2021-04-25 17:30:39 - Stored labels into file models/bird/labels.txt.
2021-04-25 17:30:39 - Train dataset size: 4338
2021-04-25 17:30:39 - Prepare Validation datasets.
2021-04-25 17:30:39 - loading annotations from: data/bird/sub-test-annotations-bbox.csv
2021-04-25 17:30:39 - annotations loaded from:  data/bird/sub-test-annotations-bbox.csv
num images:  497
2021-04-25 17:30:40 - Dataset Summary:Number of Images: 497
Minimum Number of Images for a Class: -1
Label Distribution:
	Bird: 695
2021-04-25 17:30:40 - Validation dataset size: 497
2021-04-25 17:30:40 - Build network.
2021-04-25 17:30:40 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2021-04-25 17:30:41 - Took 0.49 seconds to load the model.
2021-04-25 17:30:51 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2021-04-25 17:30:51 - Uses CosineAnnealingLR scheduler.
2021-04-25 17:30:51 - Start training from epoch 0.
/home/user/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
Traceback (most recent call last):
  File "train_ssd.py", line 343, in <module>
    device=DEVICE, debug_steps=args.debug_steps, epoch=epoch)
  File "train_ssd.py", line 113, in train
    for i, data in enumerate(loader):
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 363, in __next__
    data = self._next_data()
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 989, in _next_data
    return self._process_data(data)
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1014, in _process_data
    data.reraise()
  File "/home/user/.local/lib/python3.6/site-packages/torch/_utils.py", line 395, in reraise
    raise self.exc_type(msg)
AttributeError: Caught AttributeError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 185, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/user/.local/lib/python3.6/site-packages/torch/utils/data/dataset.py", line 207, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/media/user/SD/jetson-inference/python/training/detection/ssd/vision/datasets/open_images.py", line 46, in __getitem__
    _, image, boxes, labels = self._getitem(index)
  File "/media/user/SD/jetson-inference/python/training/detection/ssd/vision/datasets/open_images.py", line 30, in _getitem
    image = self._read_image(image_info['image_id'])
  File "/media/user/SD/jetson-inference/python/training/detection/ssd/vision/datasets/open_images.py", line 108, in _read_image
    if image.shape[2] == 1:
AttributeError: 'NoneType' object has no attribute 'shape'

```

dusty-nv, 2021-04-26:
> File "/media/user/SD/jetson-inference/python/training/detection/ssd/vision/datasets/open_images.py", line 30, in _getitem
>     image = self._read_image(image_info['image_id'])
>   File [...]

For some reason, it is failing to load the image file.  And in your fruits example, it seems unable to load the CSV either.  I wonder if perhaps these files got corrupted somehow?

DJT777, 2021-04-26:
> For some reason, it is failing to load the image file. And in your fruits example, it seems unable to load the CSV either. I wonder if perhaps these files got corrupted somehow?

I think you are right about the corrupted files. To solve it I just ran the jetson-inference .py files from SD on my laptop. 

New issue:
I think I need to make a swap space for extra memory. Can you have a look at the terminal near the bottom and verify if a swap partition would solve that?

```
user@tx1:/media/user/SD/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --data=data/bird --model-dir=models/bird --batch-size=4 --epochs=30
2021-04-26 13:44:08 - Using CUDA...
2021-04-26 13:44:08 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/bird', dataset_type='open_images', datasets=['data/bird'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=30, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-04-26 13:44:08 - Prepare training datasets.
2021-04-26 13:44:08 - loading annotations from: data/bird/sub-train-annotations-bbox.csv
2021-04-26 13:44:08 - annotations loaded from:  data/bird/sub-train-annotations-bbox.csv
num images:  4338
2021-04-26 13:44:23 - Dataset Summary:Number of Images: 4338
Minimum Number of Images for a Class: -1
Label Distribution:
	Bird: 11316
2021-04-26 13:44:23 - Stored labels into file models/bird/labels.txt.
2021-04-26 13:44:23 - Train dataset size: 4338
2021-04-26 13:44:23 - Prepare Validation datasets.
2021-04-26 13:44:23 - loading annotations from: data/bird/sub-test-annotations-bbox.csv
2021-04-26 13:44:23 - annotations loaded from:  data/bird/sub-test-annotations-bbox.csv
num images:  497
2021-04-26 13:44:25 - Dataset Summary:Number of Images: 497
Minimum Number of Images for a Class: -1
Label Distribution:
	Bird: 685
2021-04-26 13:44:25 - Validation dataset size: 497
2021-04-26 13:44:25 - Build network.
2021-04-26 13:44:25 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2021-04-26 13:44:26 - Took 0.50 seconds to load the model.
2021-04-26 13:44:36 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2021-04-26 13:44:36 - Uses CosineAnnealingLR scheduler.
2021-04-26 13:44:36 - Start training from epoch 0.
/home/user/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/home/user/.local/lib/python3.6/site-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2021-04-26 13:45:06 - Epoch: 0, Step: 10/1085, Avg Loss: 9.6001, Avg Regression Loss 4.5722, Avg Classification Loss: 5.0279
2021-04-26 13:45:11 - Epoch: 0, Step: 20/1085, Avg Loss: 6.1991, Avg Regression Loss 2.9636, Avg Classification Loss: 3.2354
2021-04-26 13:45:19 - Epoch: 0, Step: 30/1085, Avg Loss: 5.2272, Avg Regression Loss 2.4967, Avg Classification Loss: 2.7305
2021-04-26 13:45:24 - Epoch: 0, Step: 40/1085, Avg Loss: 7.3053, Avg Regression Loss 4.0042, Avg Classification Loss: 3.3011
2021-04-26 13:45:30 - Epoch: 0, Step: 50/1085, Avg Loss: 5.3907, Avg Regression Loss 2.4520, Avg Classification Loss: 2.9387
2021-04-26 13:45:35 - Epoch: 0, Step: 60/1085, Avg Loss: 5.3235, Avg Regression Loss 2.4364, Avg Classification Loss: 2.8872
2021-04-26 13:45:43 - Epoch: 0, Step: 70/1085, Avg Loss: 5.4156, Avg Regression Loss 2.7165, Avg Classification Loss: 2.6991
2021-04-26 13:45:48 - Epoch: 0, Step: 80/1085, Avg Loss: 5.6360, Avg Regression Loss 2.7837, Avg Classification Loss: 2.8523
2021-04-26 13:45:54 - Epoch: 0, Step: 90/1085, Avg Loss: 5.2191, Avg Regression Loss 2.4698, Avg Classification Loss: 2.7494
2021-04-26 13:45:59 - Epoch: 0, Step: 100/1085, Avg Loss: 5.2288, Avg Regression Loss 2.5967, Avg Classification Loss: 2.6321
2021-04-26 13:46:06 - Epoch: 0, Step: 110/1085, Avg Loss: 4.3076, Avg Regression Loss 1.8530, Avg Classification Loss: 2.4546
2021-04-26 13:46:11 - Epoch: 0, Step: 120/1085, Avg Loss: 4.9160, Avg Regression Loss 2.3414, Avg Classification Loss: 2.5746
2021-04-26 13:46:16 - Epoch: 0, Step: 130/1085, Avg Loss: 4.8301, Avg Regression Loss 2.1542, Avg Classification Loss: 2.6758
2021-04-26 13:46:21 - Epoch: 0, Step: 140/1085, Avg Loss: 5.1814, Avg Regression Loss 2.4598, Avg Classification Loss: 2.7216
2021-04-26 13:46:29 - Epoch: 0, Step: 150/1085, Avg Loss: 4.9651, Avg Regression Loss 2.4226, Avg Classification Loss: 2.5425
2021-04-26 13:46:38 - Epoch: 0, Step: 160/1085, Avg Loss: 5.0709, Avg Regression Loss 2.6891, Avg Classification Loss: 2.3817
2021-04-26 13:46:43 - Epoch: 0, Step: 170/1085, Avg Loss: 4.5506, Avg Regression Loss 2.0874, Avg Classification Loss: 2.4632
[... cut ...]
  File "/usr/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 66, in _launch
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory

```

dusty-nv, 2021-04-26:
> OSError: [Errno 12] Cannot allocate memory

Yes, if you are running out of memory, try these suggestions:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui
* run `train_ssd.py` with `--batch-size=1` and `--workers=1`

closed 2023-04-27 by dusty-nv