dusty-nv/jetson-inference #1028: [gstreamer] gstreamer qtdemux0 ERROR This file is invalid and cannot be played

MauricioLins, 2021-04-27:
Hi @dusty-nv,

I am trying ro run the detectnet.py for a specific video to evalute my model and I am facing thi issue below, I already tried different codecs as h264 or h265 but it did not work. In this log below I have used h264 for example.

What else could I try to make it work?  This video is a mp4 file created by a NVR, I tried with some other mp4 files created using a GoPro and it has worked.

I am lost! :)

`[gstreamer] gstreamer qtdemux0 ERROR This file is invalid and cannot be played.
[gstreamer] gstreamer Debugging info: qtdemux.c(747): gst_qtdemux_pull_atom (): /GstPipeline:pipeline0/GstQTDemux:qtdemux0:
atom has bogus size 1229802312
[gstreamer] gstreamer qtdemux0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: qtdemux.c(6073): gst_qtdemux_loop (): /GstPipeline:pipeline0/GstQTDemux:qtdemux0:
streaming stopped, reason error (-5)
[gstreamer] gstreamer qtdemux0 ERROR This file contains no playable streams.
[gstreamer] gstreamer Debugging info: qtdemux.c(701): gst_qtdemux_post_no_playable_stream_error (): /GstPipeline:pipeline0/GstQTDemux:qtdemux0:
no known streams found`

Thank you for your help.

MauricioLins, 2021-04-29:
I solved it, perhaps not in the best way but it is working now.
I've recorded the video using the screen screenshot app and then I saved it as .mp4. Using this new file I am able to process the video.
Again, I am not sure if it is the best solution to be honest, but due the delivery timming that I have and the time that I was spending to analise and debug the issues, I've decided to go with this workaround.

rajeshroy402, 2022-01-31:
Faced the same thing, so @dusty-nv my video is CCTV footage, do I need to add something more in the arguments?

dusty-nv, 2022-01-31:
@rajeshroy402 I don't believe this issue is related to my code but rather GStreamer or something in the video file(s) themselves.

If you are able to find a working GStreamer pipeline that plays them, then I can look at the necessary code changes in jetson-utils.

However you may just want to transcode the video files using a tool like ffmpeg and see if GStreamer can play them then.