dusty-nv/jetson-inference #1029: Training Custom model - Issue saving model

Lyns01, 2021-04-28:
Hello,
I have been trying to train the model with custom dataset using train_ssd.py.
Training goes well but after each epoch the model gets saved with "inf" instead of the loss value in the file name. 
And therefore while converting to .onnx file , the program fails to find the model with the minimum loss.
Why is it not able to save the model with the loss value itself (as the file name) ?

dusty-nv, 2021-04-28:
Hi @Lyns01, you can manually specify the pytorch checkpoint with the `--input` flag to `onnx_export.py`

But even if you did, the model is unlikely to work well since it has INF loss.   This typically occurs because of some malformed/invalid bounding box(s) in your dataset.  Maybe some bounding box has very large or negative coordinates somehow.

What I recommend doing is to uncomment this line of code:

https://github.com/dusty-nv/pytorch-ssd/blob/e7b5af50a157c50d3bab8f55089ce57c2c812f37/vision/datasets/voc_dataset.py#L76

And then run `train_ssd.py` with `--batch_size=1 --num_workers=1 --debug_steps=1`
And then when you see the INF/NaN loss first appear, look at the annotations that were loaded just above.  Check the XML file to make sure they are ok.

You can also post the console log here when you run it as described above.

Lyns01, 2021-04-29:
Hello,
I did uncomment the print() statement and trained once again using train_ssd.py with batch size, workers and debug steps as mentioned above.

I can see every image being trained and therefore could see that one image gives "inf" in the place of loss. But couldn't find which image is causing the trouble as it doesn't print the image name. 
I have attached the console image

Lyns01, 2021-04-29:
Hi,
I did everything as described above.

Was able to notice that one image gives "inf" in the place of loss value. But couldn't find which image becoz it doesn't print the image name even after uncommenting the print statment as u said.

Have shared a screenshot of the console window
[image]

dusty-nv, 2021-04-29:
> But couldn't find which image becoz it doesn't print the image name even after uncommenting the print statment as u said.

Are you running this in the container by chance?

If so, you need to edit the version of the file inside the container.  You can do this by starting the container, and then editing the file inside the container (with a text editor like nano).  If text editor isn't installed, you can install it into container by running `apt-get update && apt-get install nano`

Lyns01, 2021-05-04:
Hi yes I was running in docker. And yes there was a problem with the image annotations in the dataset. I tried again with a proper data set and everything works fine. Thank you for helping me sort it out.

closed 2021-05-04 by Lyns01