dusty-nv/jetson-inference #1031: jetson.inference.detectNet waiting time of 5 minutes

yV11, 2021-04-29:
Hey there, 

We use the following code line `self.net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)` and it takes about 5 minutes to load.

Is there a way to speed up the process or to preload the model if we start the robot?

dusty-nv, 2021-04-29:
It should only take that long the first time you run the model, because it then saves the optimized TensorRT engine out to disk.

Do you see the .engine file get saved in the same directory of the model?  It should be a *.engine file under `jetson-inference/data/networks/ssd-mobilenet-v2`

yV11, 2021-04-30:
- No there is no *.engine file in this destination
- There are only the two following files: ssd_coco_labels.txt and ssd_mobilenet_v2_coco.uff

dusty-nv, 2021-04-30:
Do you have write access to the folder?

If you run `detectnet` program outside of Jupyter notebook, does it make the .engine file?

yV11, 2021-05-03:
- Thanks @dusty-nv , there was a problem with the permissions for this folder & files, after changing the permission the `*.engine` file has been created and the loading time was really fast

closed 2021-05-03 by yV11