dusty-nv/jetson-inference #1032: Question on training with custom dataset

Lyns01, 2021-05-04:
Hello,
I have 4 voc dataset (each with ~ 200 images and their annotations). Each dataset has images of a particular object. So 4 different objects 4 different datasets. 
Now how do I train my model?
Should I try merging everything into one? Or is there a way to train it one by one? 
I also came across --resume in train_ssd.py where we can collect more data, come back and start training from where we left with the .pth file of the model created previously. 
But that doesn't help if I am going to train it with a new object and hence a new label.

dusty-nv, 2021-05-04:
Hi @Lyns01, what I would do is merge them into one dataset.  Fortunately the VOC format should make this fairly easy, as long as you have different image filenames/annotation filenames.  You can simply put all the images under `JPEGImages/` and annotations under `Annotations/`. 

Then you basically would take your different ImageSets under `ImageSets/Main` and combine the 4 separate files together.   For example, take `ImageSets/Main/trainval.txt` from the 4 separate datasets and combine them into one `ImageSets/Main/trainval.txt` (and do that for `test.txt`, `train.txt`, `trainval.txt`, and `val.txt`)

Lyns01, 2021-05-06:
Thanks for you reply.. I have one more question, I have a dataset of 300 images (one object only).. I trained them for 30 epochs (batch size 4 and workers 2).. at the end the loss value was around 0.8 
I converted it to .onnx file and tested using detectnet..
I was able to detect the actual object but in addition draws bounding boxes around every object it captures.. 

Can u suggest on how to make it a better model? 
Should I increase the number of images or epochs or wat should be done?

dusty-nv, 2021-05-06:
> Should I increase the number of images or epochs or wat should be done?

Try doing both.  Also in your images, have the background objects that you don't want to detect in the images - just don't annotate those unwanted/background objects.

Also if you train your model again, delete the .engine file in your model's directory, otherwise TensorRT will use old model.

Lyns01, 2021-05-07:
During each epoch there are three loss values being displayed. Validation regression loss, Validation classification loss, and validation loss. Is there a way to print the training loss too??

Lyns01, 2021-05-07:
and is there any ways to augment the dataset?

Lyns01, 2021-05-08:
Also wat can be the size of each image in the dataset? If the image size is around 5 mb, the process gets killed..

dusty-nv, 2021-05-08:
What Jetson are you using?  If you are on Nano see the sections of the tutorial about SWAP and disabling the desktop GUI.  Also set --batch-size 1 --num-workers 1

I believe the dataset already gets augmented with additional transformations to add to the data.

The training loss displays during each N batches, which you can set with --debug-steps argument

Lyns01, 2021-05-13:
Thanks dusty.

closed 2021-05-13 by Lyns01