dusty-nv/jetson-inference #1033: Python examples import error in Docker r32.5.0 dustynv/jetson-inference

De-Ma, 2021-05-04:
Hi,

i have troubles running any python code in the Docker Container r32.5.0 dustynv/jetson-inference. 
I followed the instrucitons here: 
https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md
```
$ git clone --recursive https://github.com/dusty-nv/jetson-inference
$ cd jetson-inference
$ docker/run.sh
```
after running the run.sh script i can see the Docker running and attached VS-Code to it.
Then i tried to run any python example but it seems that it always crashes/exception when importing the first library (with Pyhton2 and Pyhton3):

```
> Executing task: docker exec -it a81f057915a28cdd8b106532c8584083b331ad160e942ece3151c004872bd1b2 bash <
root@##:/jetson-inference# cd python/
root@##:/jetson-inference/python# ls
CMakeLists.txt  bindings  examples  python  training
root@##:/jetson-inference/python# cd examples/
root@##:/jetson-inference/python/examples# ls
README.md  detectnet.py  imagenet.py  my-detection.py  my-recognition.py  segnet.py  segnet_utils.py
root@##:/jetson-inference/python/examples# python3 my-detection.py 
Traceback (most recent call last):
  File "my-detection.py", line 24, in <module>
    import jetson.inference
  File "/usr/lib/python3.6/dist-packages/jetson/inference/__init__.py", line 5, in <module>
    import jetson.utils
  File "/usr/lib/python3.6/dist-packages/jetson/utils/__init__.py", line 4, in <module>
    from jetson_utils_python import *
ImportError: libnppicc.so.10: cannot open shared object file: No such file or directory
root@##:/jetson-inference/python/examples# python my-detection.py 
Traceback (most recent call last):
  File "my-detection.py", line 24, in <module>
    import jetson.inference
  File "/usr/lib/python3.6/dist-packages/jetson/inference/__init__.py", line 5, in <module>
    import jetson.utils
  File "/usr/lib/python3.6/dist-packages/jetson/utils/__init__.py", line 4, in <module>
    from jetson_utils_python import *
ImportError: libnppicc.so.10: cannot open shared object file: No such file or directory
root@##:/jetson-inference/python/examples# 
```
My system info is:
```
        NVIDIA Jetson Nano (Developer Kit Version)
        L4T 32.5.0 [ JetPack UNKNOWN ]
        Ubuntu 18.04.5 LTS
        Kernel Version: 4.9.201-tegra
        CUDA NOT_INSTALLED
        CUDA Architecture: 5.3
        OpenCV version: 4.1.1
        OpenCV Cuda: YES
        CUDNN: NOT_INSTALLED
        TensorRT: NOT_INSTALLED
        Vision Works: NOT_INSTALLED
        VPI: NOT_INSTALLED
```

dusty-nv, 2021-05-04:
> CUDA NOT_INSTALLED
>         CUDA Architecture: 5.3
>         OpenCV version: 4.1.1
>         OpenCV Cuda: YES
>         CUDNN: NOT_INSTALLED
>         TensorRT: NOT_INSTALLED
>         Vision Works: NOT_INSTALLED
>         VPI: NOT_INSTALLED

Hi @De-Ma, you don't have the CUDA/cuDNN/TensorRT libraries installed on your Jetson device.   These are mapped into the container at runtime (for more info about that, see here:  https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-Container-Runtime-on-Jetson#mount-plugins)

You will need to install these first.  Typically these already come with the SD card image or when you flash with SDK Manager.

De-Ma, 2021-05-06:
@dusty-nv Tanks for the fast response, i installed the whole packages neccessary with the SDK-Manager on my running system and now everything seems to work fine now. 

Great thanks for your help.
```
	NVIDIA Jetson Nano (Developer Kit Version)
	L4T 32.5.0 [ JetPack UNKNOWN ]
	Ubuntu 18.04.5 LTS
	Kernel Version: 4.9.201-tegra
	CUDA 10.2.89
	CUDA Architecture: 5.3
	OpenCV version: 4.1.1
	OpenCV Cuda: YES
	CUDNN: 8.0.0.180
	TensorRT: 7.1.3.0
	Vision Works: 1.6.0.501
	VPI: ii libnvvpi1 1.0.12 arm64 NVIDIA Vision Programming Interface library
```

closed 2021-05-06 by De-Ma

arun-kumark, 2022-09-25:
On my Jetson baord also, everything seems installed: I could see the things below:
~/Documents/jetson-inference$ jetson_release -v

NVIDIA Jetson Xavier NX (Developer Kit Version)
Jetpack 4.6 [L4T 32.6.1]
NV Power Mode: MODE_20W_6CORE - Type: 8
jetson_stats.service: active
Board info:
Type: Xavier NX (Developer Kit Version)
SOC Family: tegra194 - ID:25
Module: P3668 - Board: P3509-000
Code Name: jakku
CUDA GPU architecture (ARCH_BIN): 7.2
Serial Number: 1425020057815
Libraries:
CUDA: 10.2.300
cuDNN: 8.2.1.32
TensorRT: 8.0.1.6
Visionworks: 1.6.0.501
OpenCV: 4.6.0 compiled CUDA: YES
VPI: ii libnvvpi1 1.1.15 arm64 NVIDIA Vision Programming Interface library
Vulkan: 1.2.70
jetson-stats:
Version 3.1.4
Works on Python 3.6.9
~/Documents/jetson-inference$
But when I run the container, it says
./video-viewer: error while loading shared libraries: libnppicc.so.10: cannot open shared object file: No such file or directory

Looks to me its related to CUDA not installed.

Could you solve it or hint how it was solved for you ?

thanks

dusty-nv, 2022-09-25:
Hi @arun-kumark, can you see these libraries on your device outside of the container?

``` bash
$ ls -ll /usr/local/cuda/lib64/libnppicc*
lrwxrwxrwx 1 root root      15 Mar  1  2021 /usr/local/cuda/lib64/libnppicc.so -> libnppicc.so.10
lrwxrwxrwx 1 root root      23 Mar  1  2021 /usr/local/cuda/lib64/libnppicc.so.10 -> libnppicc.so.10.2.1.300
-rw-r--r-- 1 root root 4914920 Mar  1  2021 /usr/local/cuda/lib64/libnppicc.so.10.2.1.300
-rw-r--r-- 1 root root 5722536 Mar  1  2021 /usr/local/cuda/lib64/libnppicc_static.a
```

You could also just try building the project from source:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md
If CUDA/TensorRT is indeed installed on your Xavier NX, that should take just a few minutes.