dusty-nv/jetson-inference #1034: Transfering learning on custom dataset

priyaganaboor-hash, 2021-05-05:
I have collected the images of red geranium plant and red azalea plants. Than, I have trained them on alexnet, vgg16 and resnet50. Now, I have converted them onnx and to TensorRT. The whole process it as different class. 
For example: The vgg16 classifies as,
class 0580 - 0.045837  (greenhouse, nursery, glasshouse)
class 0706 - 0.031021  (patio, terrace)
class 0738 - 0.818848  (pot, flowerpot)
class 0883 - 0.018097  (vase)

How can I test correct classification as only azalea or geranium.

dusty-nv, 2021-05-05:
Hi @priyaganaboor-hash, is your labels.txt listed in alphabetical order?

What does the folder structure of your dataset look like?

What accuracy did PyTorch train your model to?

priyaganaboor, 2021-05-05:
Hi,

>>Yes, The labels.txt listed in alphabetical order.

>>The folder structure of dataset is ,
‣ train/
	• azalea/
	• geranium/
‣ val/
	• azalea/
	• geranium/
‣ test/
	• azalea/
	• geranium/

>>For all the models I got accuracy between 70 to 80%.

>>I have trained the network on my local pc, on a custom program, and even converted to .onnx format, in my local PC.  After that, I am running imagenet.py(from the repo) to convert it to TensorRT and display the single image classification on Jetson nano.

dusty-nv, 2021-05-05:
What does your labels.txt look like?

Are you specifying `--labels=<my-labels-file.txt>` on the command line?  Because it appears to be using the default labels.txt from 1000-class imagenet model.

What command line are you using to run the model in imagenet/imagenet.py?

priyaganaboor, 2021-05-05:
Yes, I am mentioning the --labels in the command-line argument.

NET=models/PlantClassification
DATASET=data/PlantClassification

$ imagenet.py --model=$NET/PlantClassification/vgg16.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/geranium/test_img_01.jpg  $DATASET/test/geranium/output.jpg

I didn't understand --input_blob=input_0 --output_blob=output_0  so I tried with and that command thinking image name format.

priyaganaboor, 2021-05-05:
@dusty-nv, I  have used  -- network=vgg16, is this correct? I saw the default is googlenet so changed it.

dusty-nv, 2021-05-05:
Try it without `--network`, like the following:

``` bash
imagenet.py --model=$NET/vgg16.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt $DATASET/test/geranium/test_img_01.jpg $DATASET/test/geranium/output.jpg
```

priyaganaboor-hash, 2021-05-06:
Hey @dusty-nv 

for command,
$ imagenet.py --model=models/PlantClassification/vgg.onnx --input_blob=input_0 --output_blob=output_0 --labels=data/PlantClassification/labels.txt data/PlantClassification/test/azalea/test5.jpeg data/PlantClassification/test/azalea/output.jpeg

I am getting error as,
[TRT]    device GPU, models/PlantClassification/vgg.onnx initialized.
[TRT]    imageNet -- loaded 3 class info entries
[TRT]    imageNet -- didn't load expected number of class descriptions  (3 of 2)
[TRT]    imageNet -- failed to load synset class descriptions  (3 / 3 of 2)
[TRT]    imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 55, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network

I have only two classes, I dont understand this error.

priyaganaboor-hash, 2021-05-06:
Hey! @dusty-nv 

I just deleted labels.txt  and created new one and used the same above command again, It worked this time.

Thank you so much! You have done a great job.

closed 2021-05-06 by priyaganaboor-hash