dusty-nv/jetson-inference #1035: Inference with jetson.utils.imageNet() not achieving the same precision as Pytorch built code.

Andres-celi, 2021-05-06:
Hello everyone,

I have a resnet152 model. When I try to do inference with it on pytorch. It works great.

For example:
`for key in imgs:
            self.img = imgs[key]
            print(self.img)
            self.height,self.width = self.img.shape[:2]
            print(self.img.shape[:])
            self.imagen = self.img[:,int(self.width*0.1):int(self.width*0.85)]
            print(self.imagen.shape[:])
            self.data = Image.fromarray(cv2.cvtColor(self.imagen, cv2.COLOR_BGR2RGB))
            self.test_transforms = transforms.Compose([transforms.Resize(270),
                                      transforms.CenterCrop(224),                                      
                transforms.ToTensor(),
                self.normalize,
                                     ])
            self.image_tensor = transforms.functional.adjust_brightness(self.data,1)
            self.image_tensor = transforms.functional.adjust_saturation(self.image_tensor,1)
            self.image_tensor = self.test_transforms(self.image_tensor).float()
            print(self.image_tensor.shape[:])
            self.images.append(self.image_tensor)
        print(self.images[0])
        self.images_dataset = MyDataset(self.images)
        self.loader = DataLoader(self.images_dataset,batch_size=len(self.images),num_workers=1,pin_memory=True,shuffle=False)
        with torch.no_grad():
            for i,images in enumerate(self.loader):
                images=images.cuda(0,non_blocking=True)
                #t0 = time.time()
                prediction = self.model(images)
                topk=(1, min(5, len(self.objetos)))
                maxk = max(topk)
                val, pred = prediction.topk(maxk, 1, True, True)
                pred = pred.t()
                ids = pred.cpu().numpy()[:]
                sm = torch.nn.Softmax(dim=-1)
                #print("La prediccion es: ",val, sm(val))
                val = sm(val).t()            
                porcentajes = val.cpu().detach().numpy()[:]`

but when I use the jetson.inference.ImageNet() in the imagenet.py code given by this repository. The results are very different.

Does anybody have an idea of why?

My impression is that the difference comes from the way I manipulate the images with torchvision.transforms() or basically this:

` self.test_transforms = transforms.Compose([transforms.Resize(270),
                                      transforms.CenterCrop(224),                                      
                transforms.ToTensor(),
                self.normalize,
                                     ])`
                                     
But if anybody has any ideas please I would love to hear them. 

Thank you all in advance.

dusty-nv, 2021-05-06:
The ONNX codepath in imagenet is setup to use the same normalization constants from this training script:

https://github.com/dusty-nv/pytorch-classification/blob/15c7d51cd6a1271e5ee6959c3f91fb95b6b54ff0/train.py#L161

Here is where the pre-processing is done for imagenet:

https://github.com/dusty-nv/jetson-inference/blob/2fb798e3e4895b51ce7315826297cf321f4bd577/c/imageNet.cpp#L433

It applies the normalization using the constants, and then downscales to 224x224.   So if you have additional/different pre-processing required you would need to also make these changes to get similar results.

Andres-celi, 2021-05-06:
Dang it... it seems party similar... thanks for the quick response. I actually did use the original train.py you linked to train my resnet152 and based on that did the inference code too so I am kind of out of ideas. In your experience the imagenet.py always performs pretty much the same as straight up pytorch?

dusty-nv, 2021-05-06:
Yes, and you may want to delete the .engine file in your model's directory if you have exported to ONNX multiple times (otherwise TensorRT could be using old model)

Are your normalization constants the same?   You are also doing these transforms:

``` python
self.image_tensor = transforms.functional.adjust_brightness(self.data,1)
self.image_tensor = transforms.functional.adjust_saturation(self.image_tensor,1)
```

dusty-nv, 2021-05-06:
It looks like you are center-cropping the test images in PyTorch whereas at runtime I simply resize then to 224x224.

When you say the results are very different, can you clarify what you mean?   Is it the classification accuracy or do you mean the probabilities are very different?   

Can you double-check that your labels.txt is in alphabetical order?

Andres-celi, 2021-05-06:
Yes, when I use imagenet() I actually try to resize and input an already centercropped 224x224 image:

	`img = jetson.utils.loadImage(os.path.join(path,img))`
	`print('RGB image: ')`
	`print(img)`
	`img = jetson.utils.cudaToNumpy(img)`
	
	`height,width = img.shape[:2]`
	`print(img.shape[:])`
	`img = img[:,int(width*0.1):int(width*0.85)]`
	`print(type(img),img.shape[:],np.amax(img),np.amin(img))`

	`img = Image.fromarray(img)`

	`size=(270,270)`
	`img = img.resize(size)`
	`print(img.size)`
	`print(type(img))`
	`#height,width = img.shape[:2]`
	`height,width = img.size`

	`left = 23`
	`top = 23`
	`right = 247`
	`bottom = 247`

	`# Cropped image of above dimension `
	`# (It will not change orginal image) `
	`img = img.crop((left, top, right, bottom))`

	`print(img.size)`

	`img = np.array(img)`

	`#img1 = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)`
	
	`print(type(img))`
	`img = jetson.utils.cudaFromNumpy(img)`
	`# classify the image`
	`class_idx, confidence = net.Classify(img)`
	
My results are different in confidence and the actual prediction at times. I hace approx 200 images processed with torch, when I run my imagenet() it gets the same prediction 50% of the times and almost never the same confidence.

labels.txt is in alphabetical order, already had my first scare with that. Lol.

Andres-celi, 2021-05-06:
Maybe can you point me towards an example of proper CudaImage manipulation? to use:

`img = jetson.utils.loadImage(os.path.join(path,img))`

or

`img = input.Capture()`

and then not leave the CudaImage format until I call the Classify() function?

Andres-celi, 2021-05-06:
I will check that out.

When I tried the other resnet architecture they could not achieve the requiered accuracy so I haven´t played with them in a while.

But I you say that I should get the same with the jetson-inference library I will keep trying.