dusty-nv/jetson-inference #1040: detectnet with video files

MauricioLins, 2021-05-10:
Hi all,

I have trained a new model and I also have customised the detecnet.py script to work as expected in my use case. Every time that I run it using live (using a camera) it works perfectly. 

The issue is when I try to run the same for video files, it does not work properly and I can't identify why. What is wierd is that even using the same videos that I used to train the model it does not work.

I started thinking that perhaps there is something else that I need to do to address the video processing, because as I mentioned I've tried many different approaches for video files and for all those tries the detection did not work.

Does anyone have any idea?

dusty-nv, 2021-05-11:
Hmm.  So it runs fine on the live camera, but on the recorded video file of the same live camera, the detections are much lower?

Are you able to record the camera's video file at a higher bitrate?

MauricioLins, 2021-05-11:
Oh sorry, I've missed this important information.
It's been working with usb or csi camera, but it does not work with the videos that I have from the cctv camera.
I've also used these videos in my training data and I am testing all scenarios now.
I am afraid I will need more data to train for this specific camera, I don't know

dusty-nv, 2021-05-11:
> The point is, I need to think how to create this logic here using the this detectnet approach. @dusty-nv, have you had the opportunity to create something with similar approach?

I personally haven't, but I recall another person who had use detectNet first and then imageNet to classify the results.  It shouldn't be an issue to combine their use in one script.  Although for more advanced application you may want to use DeepStream and Transfer Learning Toolkit to train your model(s)

closed 2023-04-27 by dusty-nv